spacetelescope/jwql #437: API ReadTheDocs failing

laurenmarietta, 2019-06-05:
It seems as though our webhook to ReadTheDocs is not currently updating the API docs:
[image]

Actually, it looks like our doc builds have been failing for a month!
https://readthedocs.org/projects/jwql/builds/

There is an error message here that should be helpful for investigating: https://readthedocs.org/projects/jwql/builds/9177249/

bourque, 2019-07-24:
@catherine-martlin I made just a few tweaks (#474) to get things working.  The main problem actually was that RTD was pointing to the `master` branch, and some of the fixes were simply just not merged into `master` yet (e.g. #463).  I went into the advanced RTD settings to switch the default branch to `develop` and builds started passing.

I think we should probably just keep pointing to `develop` branch anyways since that is what we mostly use right now.  Perhaps we can switch back to `master` after our 1.0 release and after we actually have a user base.

catherine-martlin, 2019-07-26:
I agree - it definitely makes more sense to point to develop until we have a 1.0 release for master!

PR #474 (2019-08-06): Update API docs
This PR provides a small update to the `sphinx` build configuration and API docs to get them up to date.

Closes #437 

!Please do not delete this branch once merged!
closed 2019-08-06 by bourque