spacetelescope/jwql #446: Filename parser crashed on capital MIRIMAGE

cracraft, 2019-06-21:
I had just put in some dark MIRI files created through the jwst pipeline script create_data for testing the dark current monitor, and the filenames were in this format: jw04191001001_01101_00001_MIRIMAGE_uncal.fits. I then tried to continue testing the filesystem monitor, and it crashed on the new files due to the MIRIMAGE being capital case. The error showed up in the filename parser code. (showed up in log monitor_filesystem_2019-06-21-10-17.log in the dev directory) Once I changed the capital to lowercase file names, the script ran fine, and filename parser was happy. Do we need to make sure the parser can handle unexpected cases, or just change our test data to make sure it's in the expected format?

bourque, 2019-06-24:
I believe that the convention that the archive employs (and will continue to employ in the future) based on their current documentation and what was given to us for a filesystem is that these filenames will be lowercase.  My opinion on this is that `filename_parser` should continue to enforce that standard since that is the expectation.  

@laurenmarietta Do you have any opinions on this?

laurenmarietta, 2019-06-25:
I agree that `filename_parser` should match the expected formal naming schema, and not necessarily be modified to work for one-off exceptions. But since I imagine the naming schema is still somewhat in flux, we should be sure to make sure we're up to date with the latest version.

@cracraft was it a version of the pipeline produce a file with a capitalized name? As in, has the pipeline been changed recently to capitalize the names? If that is the case, then maybe we should accommodate it.

cracraft, 2019-06-25:
This was a product that I had from CV3 run through a pipeline piece of code called create_data. That code was meant to produce pipeline ready for testing files, but has had bugs and errors along the way, so the capitalization is older code, not updated to match more recent filename schemes. My question wasn't about whether this was the expected naming scheme, just whether we wanted the parser to be robust to test names, or be more strict to what is expected in flight. If we want to keep it strict for flight, then we can go ahead and close the issue.

bourque, 2019-06-25:
Thanks @cracraft and @laurenmarietta.  We will close this issue for now, but let's stay cognizant of potential changes to filename conventions coming out of the pipeline.

closed 2019-06-25 by bourque