spacetelescope/jwql #467: Dashboard blank on web server

laurenmarietta, 2019-07-15:
We are seeing a repeat of the error documented in #225, in which the dashboard does not load on the web server. The following error appears in the console: 
```
GET https://dljwql.stsci.edu/dashboard/ net::ERR_CONTENT_LENGTH_MISMATCH 200 (OK)
```

[StackOverflow](https://stackoverflow.com/questions/22183859/javascript-err-content-length-mismatch) suggests this is due to a permissions problem with `nginx`. However, none of us have permissions to do anything with `nginx` on the server, so in #225 this error was fixed by @alexyermolaev.

However, since this problem has now arisen twice, I think it would be wise to either (1) develop a more permanent solution so this permissions error won't happen again or (2) make changes so that we are capable of fixing it ourselves (reducing single points of failure).

bourque, 2019-09-16:
Ok I think I've figured out the issue here.  The various `.html` and `.js` files that are created by `monitor_filesystem.py`, `monitor_mast.py`, and `monitor_cron_jobs.py` contained references to `bokeh` version `1.3.2` or `1.2.0`.  I think this is because the `bokeh` versions installed on the dev server were out of date, and so when these files were re-written by the execution of scripts via `cron`, they were using the old `bokeh` version.

After updating the `bokeh` version on the dev server, and re-generating the `.html`/`.js` files, everything seems to be rendering now.

I am currently regenerating the `filesystem_monitor` outputs, but those take a while so they will not appear on the dashboard right away.

bourque, 2019-09-16:
Another note: I was only able to figure out this problem by looking through the console messages in FireFox developer tools.  The Chrome console reported cryptic error messages (i.e. `Line Length Mismatch`, while Firefox made it apparent that there was a mismatch with `Bokeh` versions.

bourque, 2019-09-16:
I think moving forward, we just have to be aware that when we update  `bokeh` in our environment files and `requirements.txt`, we should also be updating the installed version on the dev server.  Currently, we have a `cron` job to do this for the `conda` environment used for serving the web app, but not the `conda` environment used for running scripts via `cron`.  I will add a `cron` job to take care of this automatically.

bourque, 2019-09-17:
Turns out that the `bokeh` version mismatch was not the only problem.  I am also seeing these errors come up on the console.  Currently not sure how to fix this.

[image]

bourque, 2020-10-20:
This will be solved with #641

closed 2021-04-01 by bourque