spacetelescope/jwql #473: Add "dragon's breath" as an anomaly

bhilbert4, 2019-07-23:
The NIRCam team spent time at a meeting today talking about dragon's breath, which is scattered light on the detector in the case where a star lands just outside the detector on the edge of the detector mask. This scatters some fraction of the star's light onto the detector.

From the discussion it's clear that this is something we want to track and identify in NIRCam data. So we should add this as a specific option in the anomalies variable in utils/constants.

bourque, 2019-07-23:
Perhaps we should take this opportunity to build the infrastructure to support instrument-specific anomalies via their own classes or subclasses in `database_interface`

bourque, 2020-12-18:
I think this is solved now but somehow this issue fell through the cracks.  I will close this.

closed 2020-12-18 by bourque