spacetelescope/jwql #491: Isilon filesystem contains un-parsable filenames

bourque, 2019-08-19:
Running `test_filename_parser_whole_filesystem` while pointing to the filesystem on isilon storage reveals the existence of some filenames that can't be parsed.  A few examples are provided below:

```
jw00629-o003_t004_nircam_f212n-nrcb1_wfscmb-01.fit
jw98013001001_02106_00001_mirifulong_original.fits
jwst_miri_mask_0021.fits
jwst_nircam_flat_0299.fits
jwst_miri_saturation_0023.fits
jwst_nirspec_superbias_0102.fits
jw00625-c1008_t001_niriss_f430m-nrm-sub80_psf-amiavg.fits
```

We should determine if these types of files are intended to be in this filesystem (and if so, for how long), and if these files are something we want to support in the `filename_parser`.

cracraft, 2019-08-19:
Well, several of those are reference files (jwst_miri_mask_0021.fits) and the like, while at least the mirifulong one seems to have a tweaked version of a create_data run miri data file. There won't be any files with *original* coming out of the actual pipeline.

bourque, 2019-08-19:
Thanks for that info @cracraft

bourque, 2019-09-16:
I sent an email off to the archive asking about some of these.

After looking at the various documentation out there (namely [this](https://innerspace.stsci.edu/pages/viewpage.action?pageId=94092600) and [this](https://innerspace.stsci.edu/pages/viewpage.action?spaceKey=SCSB&title=JWST+Science+Data+Products)), it appears that some of these are Level 3 data products (e.g. `jw00625-c1008_t001_niriss_f430m-nrm-sub80_psf-amiavg.fits`). I think it is safe to say that they are intended to be in the filesystem and we should support them with `filename_parser`.  I will create a separate GitHub issue for adding support for those.

bourque, 2020-12-18:
I forget if this is still an issue or not.  We should re-run `test_filename_parser_whole_filesystem` and see if any errors come up and then deal with them accordingly.

bourque, 2021-01-06:
@bsunnquist I'm wondering if you would like/are able to take this over for the `v0.26` sprint? Should be as simple as running `test_filename_parser_whole_filesystem` and dealing with any issues that arise from it (hopefully there are none!)

bsunnquist, 2021-01-13:
@bourque sounds good!

bourque, 2021-01-14:
@bsunnquist Awesome, thank you!

bsunnquist, 2021-02-11:
Like Misty noted, some of those are reference files which shouldn't be a problem. However, those WFS&C and AMI names are legitimate - it looks like they've since been removed from our JWQL filesystem, so `test_filename_parser_whole_filesystem` passes with no changes to the code. Though, since it will eventually be an issue once we sync up with MAST, I've added support for them in this PR https://github.com/spacetelescope/jwql/pull/696

closed 2021-02-24 by bourque