spacetelescope/jwql #502: Complete Application Security Assessment

bourque, 2019-09-04:
OWASP has a way to assess the security of an application, which has been detailed in [this innerspace page](https://innerspace.stsci.edu/display/JWQLPROJ/Copy+of+Application+Security+Assessment+Template+V4.0).  We should go through this as best we can and see where we can improve.

bsunnquist, 2021-04-13:
Perhaps related, I am currently storing full central storage paths in the database tables for the bias/readnoise monitors - is this something I should change as part of this security sweep, or is the database protected?

bourque, 2021-04-21:
@bsunnquist Yes, I would advise to not store the *full* path to the items, but storing the path reletative to the main directories listed in the config file should be fine.  For example, appending something like `bias_monitor/some_dir/some_file.txt` to what is returned by `utils.get_config()['outputs']` would hide the full path to the file in the `outputs` directory in central storage.

bsunnquist, 2021-04-21:
@bourque sounds good - I can make a PR with those changes and link it here

bourque, 2021-08-09:
This issue might actually not be needed to be resolved.  Since the production web app is going to be behind a SSO layer and only available to internal instrument team members, we need not worry so much.  I will close this for now, but if ITSD ends up requiring this we can re-open it.

closed 2021-08-09 by bourque

mfixstsci, 2021-08-09:
🤞 Let's hope so, this list is pretty long.