spacetelescope/jwql #504: Get tests running on Internal Jenkins server

bourque, 2019-09-10:
We currently have several tests that require access to central storage and thus cannot be run on our existing Jenkins server, since it does not have access (for example, [this test](https://github.com/spacetelescope/jwql/blob/develop/jwql/tests/test_utils.py#L276)).  Currently, we avoid running these tests altogether if they are executed from the Jenkins server (i.e. if `ON_JENKINS` is `True`).

DATB has set up a separate Jenkins server that does have access to central storage (#144), however currently I don't believe we actually have anything building/testing on it.

We should try to get tests running (and passing!) on this internal Jenkins server, and make the results available somehow. I believe the [`JenkinsfileRT`](https://github.com/spacetelescope/jwql/blob/develop/JenkinsfileRT) file contains the code to run things on there.

jbcurtin, 2019-09-11:
We have a few different JENKINS runtimes, I'd like to change ON_JENKINS to an enum. What do you think?

```
class TestingEnviornemnt:
  CentralStorage: str = 'central'
  Dev: str = 'dev'
  Nightly: str = 'nightly'

@pytest.mark.skipif(ON_JENKINS == TestEnvironment.CentralStorage, reason='Requires access to central storage')
def test_make_log_file():
    ...logic...

```

bourque, 2019-09-11:
@jbcurtin Sounds good to me!

bourque, 2021-04-01:
We have now removed our use of Jenkins altogether and replaced with GitHub Actions (#698), so I don't think we need to explore this any further.

closed 2021-04-01 by bourque