spacetelescope/jwql #514: Clean up Parameter and Return values in API docs

bhilbert4, 2019-10-19:
Currently, the documentation style used to denote Parameter and Return options is not rendering quite right on Readthedocs. There are no spaces between the parameter names and types, making them difficult to read. 

For example, from [the edb documentation](https://jwql.readthedocs.io/en/stable/edb.html#jwql.edb.engineering_database.get_mnemonic), there is the `mnemonic_identifier` parameter, which is a string, is listed as: `mnemonic_identifierstr`.

We should figure out what documentation format Readthedocs is looking for and update our documentation to match.

bhilbert4, 2019-10-19:
Looking at examples of projects' documentation built using sphinx's rtd theme, it looks like we would need to do something like that done for [arcade](http://arcade.academy/arcade.html#arcade.create_orthogonal_projection):

```
def create_orthogonal_projection(
        left,
        right,
        bottom,
        top,
        near,
        far,
        dtype=None
):
    """
    Creates an orthogonal projection matrix. Used internally with the
    OpenGL shaders.

    :param float left: The left of the near plane relative to the plane's centre.
    :param float right: The right of the near plane relative to the plane's centre.
    :param float top: The top of the near plane relative to the plane's centre.
    :param float bottom: The bottom of the near plane relative to the plane's centre.
    :param float near: The distance of the near plane from the camera's origin.
                       It is recommended that the near plane is set to 1.0 or above to avoid
                       rendering issues at close range.
    :param float far: The distance of the far plane from the camera's origin.
    :param dtype:
    :return: A projection matrix representing the specified orthogonal perspective.
    :rtype: numpy.array

    .. seealso:: http://msdn.microsoft.com/en-us/library/dd373965(v=vs.85).aspx
    """
```

Here's another example from [flake8](http://flake8.pycqa.org/en/latest/user/python-api.html#flake8.api.legacy.StyleGuide.input_file). The source used for this page is:

```
def input_file(self, filename, lines=None, expected=None, line_offset=0):
        """Run collected checks on a single file.

        This will check the file passed in and return a :class:`Report`
        instance.

        :param str filename:
            The path to the file to check.
        :param list lines:
            Ignored since Flake8 3.0.
        :param expected:
            Ignored since Flake8 3.0.
        :param int line_offset:
            Ignored since Flake8 3.0.
        :returns:
            Object that mimic's Flake8 2.0's Reporter class.
        :rtype:
            flake8.api.legacy.Report
        """
        return self.check_files([filename])
```

bourque, 2019-10-21:
Interesting.  That would be quite an overhaul if we had to change everything to have `:param` and `:returns`.  We should confirm that our docs setup is actually using the `numpydoc` convention.

bhilbert4, 2020-11-10:
This issue seems to have resolved itself. The documentation now looks as expected, with spaces and a colon between parameter names and types.

closed 2020-11-10 by bhilbert4