spacetelescope/jwql #553: Build Cosmic Ray Monitor

bourque, 2020-03-23:
Currently the MIRI and NIRCam teams are interested in this.

bhilbert4, 2020-03-23:
Knowledge of the exact rate of CR hits would be useful for telling users what an effective maximum exposure time should be. 

My thought would be that incoming data is run through calwebb_detector1 pipeline steps through the jump step (turning off CR-neighbor flagging). Counting up the number of CR impacted pixels, as well as the number of CR hits (not sure how to do this) per time would be useful. Monitoring these rates over time and correlating with solar phase would be interesting.

bhilbert4, 2021-11-09:
A late request from the NIRCam team for the Cosmic Ray monitor: would it be possible to measure the signal associated with the cosmic rays and produce intensity histograms?

closed 2022-09-06 by mfixstsci