spacetelescope/jwql #617: Readnoise Monitor crashing with pysaif issue

bourque, 2020-08-24:
```
Working on aperture NRCA5_FULL_MASKLWB_F444W in nircam
08/24/2020 11:42:16 AM INFO: 	No query history for NRCA5_FULL_MASKLWB_F444W. Beginning search date will be set to 57357.0.
08/24/2020 11:42:16 AM INFO: 	Query times: 57327.0 59085.62589506286
08/24/2020 11:42:17 AM INFO: 	Aperture: NRCA5_FULL_MASKLWB_F444W, new entries: 0
08/24/2020 11:42:17 AM INFO: 	Readnoise monitor skipped. 0 new dark files for nircam, NRCA5_FULL_MASKLWB_F444W.
08/24/2020 11:42:17 AM INFO: 	Updated the query history table
08/24/2020 11:42:17 AM CRITICAL: Traceback (most recent call last):
  File "/home/jwqladm/repositories/jwql/jwql/utils/logging_functions.py", line 284, in wrapped
    func(*args, **kwargs)
  File "/home/jwqladm/repositories/jwql/jwql/utils/logging_functions.py", line 249, in wrapped
    func(*args, **kwargs)
  File "readnoise_monitor.py", line 528, in run
    siaf = Siaf(self.instrument)
  File "/grp/jwst/ins/jwql/envs/miniconda3/envs/jwql-3.6/lib/python3.6/site-packages/pysiaf/siaf.py", line 293, in __init__
    self.apertures = read.read_jwst_siaf(self.instrument, filename=filename, basepath=basepath)
  File "/grp/jwst/ins/jwql/envs/miniconda3/envs/jwql-3.6/lib/python3.6/site-packages/pysiaf/iando/read.py", line 390, in read_jwst_siaf
    "in {}".format(basepath))
OSError: Could not find SIAF data in /grp/jwst/ins/jwql/envs/miniconda3/envs/jwql-3.6/lib/python3.6/site-packages/pysiaf/prd_data/JWST/PRDOPSSOC-029/SIAFXML/SIAFXML
```

bourque, 2020-08-24:
@bsunnquist Do you know what might be going on here?

bsunnquist, 2020-08-24:
I haven't seen that before. Does this happen repeatedly in that same spot, or could it be a random central storage connection issue (we've been having a lot of those recently)?

bourque, 2020-08-25:
Good question.  I will try to run this a few more times and see if the error is repeatable.

bourque, 2020-08-25:
Hmm, I now am able to avoid the above error, but now after a few tries I repeatedly get a new one:

```
Working on aperture NIS_WFSS64C in niriss
08/25/2020 10:09:25 AM INFO: 	No query history for NIS_WFSS64C. Beginning search date will be set to 57357.0.
08/25/2020 10:09:25 AM INFO: 	Query times: 57327.0 59086.5887058788
08/25/2020 10:09:25 AM INFO: 	Aperture: NIS_WFSS64C, new entries: 2
08/25/2020 10:09:25 AM INFO: 	jw00781016001_02106_00001_nis_uncal.fits does not exist in JWQL filesystem
08/25/2020 10:09:25 AM INFO: 	Copied /grp/jwst/ins/jwql/filesystem/jw87600/jw87600023001_02105_00001_nis_uncal.fits to /grp/jwst/ins/jwql/outputs/readnoise_monitor/data/niriss_nis_wfss64c/jw87600023001_02105_00001_nis_uncal.fits
08/25/2020 10:09:25 AM INFO: 	Working on file: /grp/jwst/ins/jwql/outputs/readnoise_monitor/data/niriss_nis_wfss64c/jw87600023001_02105_00001_nis_uncal.fits
08/25/2020 10:09:25 AM INFO: 	Running pipeline on /grp/jwst/ins/jwql/outputs/readnoise_monitor/data/niriss_nis_wfss64c/jw87600023001_02105_00001_nis_uncal.fits
08/25/2020 10:09:26 AM DEBUG: Opening /grp/jwst/ins/jwql/outputs/readnoise_monitor/data/niriss_nis_wfss64c/jw87600023001_02105_00001_nis_uncal.fits as <class 'jwst.datamodels.ramp.RampModel'>
08/25/2020 10:09:26 AM WARNING: /grp/jwst/ins/jwql/envs/miniconda3/envs/jwql-3.6/lib/python3.6/site-packages/jwst/datamodels/util.py:174: NoTypeWarning: model_type not found. Opening /grp/jwst/ins/jwql/outputs/readnoise_monitor/data/niriss_nis_wfss64c/jw87600023001_02105_00001_nis_uncal.fits as a RampModel
  NoTypeWarning)

08/25/2020 10:09:26 AM INFO: DQInitStep instance created.
08/25/2020 10:09:26 AM INFO: Step DQInitStep running with args ('/grp/jwst/ins/jwql/outputs/readnoise_monitor/data/niriss_nis_wfss64c/jw87600023001_02105_00001_nis_uncal.fits',).
08/25/2020 10:09:26 AM INFO: Step DQInitStep parameters are: {'pre_hooks': [], 'post_hooks': [], 'output_file': None, 'output_dir': None, 'output_ext': '.fits', 'output_use_model': False, 'output_use_index': True, 'save_results': False, 'skip': False, 'suffix': None, 'search_output_file': True, 'input_dir': ''}
08/25/2020 10:09:26 AM INFO: Using MASK reference file /grp/crds/cache/references/jwst/jwst_niriss_mask_0014.fits
08/25/2020 10:09:27 AM DEBUG: ref substrt1=1, subsize1=2048, substrt2=1, subsize2=2048
08/25/2020 10:09:27 AM DEBUG: sci substrt1=1, subsize1=2048, substrt2=1985, subsize2=64
08/25/2020 10:09:27 AM INFO: Extracting mask subarray to match science data
08/25/2020 10:09:27 AM DEBUG: slice xstart=0, xstop=2048, ystart=1984, ystop=2048
08/25/2020 10:09:27 AM INFO: Step DQInitStep done
08/25/2020 10:09:27 AM INFO: SuperBiasStep instance created.
08/25/2020 10:09:27 AM INFO: Step SuperBiasStep running with args (<RampModel(4, 4, 64, 2048) from jw87600023001_02105_00001_nis_uncal.fits>,).
08/25/2020 10:09:27 AM INFO: Step SuperBiasStep parameters are: {'pre_hooks': [], 'post_hooks': [], 'output_file': None, 'output_dir': None, 'output_ext': '.fits', 'output_use_model': False, 'output_use_index': True, 'save_results': False, 'skip': False, 'suffix': None, 'search_output_file': True, 'input_dir': ''}
08/25/2020 10:09:27 AM INFO: Using SUPERBIAS reference file /grp/crds/cache/references/jwst/jwst_niriss_superbias_0113.fits
08/25/2020 10:09:27 AM DEBUG: ref substrt1=1, subsize1=2048, substrt2=1985, subsize2=64
08/25/2020 10:09:27 AM DEBUG: sci substrt1=1, subsize1=2048, substrt2=1985, subsize2=64
08/25/2020 10:09:27 AM INFO: Step SuperBiasStep done
08/25/2020 10:09:27 AM INFO: RefPixStep instance created.
08/25/2020 10:09:27 AM INFO: Step RefPixStep running with args (<RampModel(4, 4, 64, 2048) from jw87600023001_02105_00001_nis_uncal.fits>,).
08/25/2020 10:09:27 AM INFO: Step RefPixStep parameters are: {'pre_hooks': [], 'post_hooks': [], 'output_file': None, 'output_dir': None, 'output_ext': '.fits', 'output_use_model': False, 'output_use_index': True, 'save_results': False, 'skip': False, 'suffix': None, 'search_output_file': True, 'input_dir': '', 'odd_even_columns': True, 'use_side_ref_pixels': True, 'side_smoothing_length': 11, 'side_gain': 1.0, 'odd_even_rows': True}
08/25/2020 10:09:27 AM INFO: use_side_ref_pixels = True
08/25/2020 10:09:27 AM INFO: odd_even_columns = True
08/25/2020 10:09:27 AM INFO: side_smoothing_length = 11
08/25/2020 10:09:27 AM INFO: side_gain = 1.000000
08/25/2020 10:09:27 AM INFO: odd_even_rows = True
08/25/2020 10:09:27 AM DEBUG: Input exposure is a subarray readout
08/25/2020 10:09:27 AM DEBUG: Input exposure is a subarray readout
08/25/2020 10:09:27 AM INFO: Step RefPixStep done
08/25/2020 10:09:28 AM INFO: 	Pipeline complete. Output: /grp/jwst/ins/jwql/outputs/readnoise_monitor/data/niriss_nis_wfss64c/jw87600023001_02105_00001_nis_uncal_refpix.fits
08/25/2020 10:09:28 AM INFO: 	Amplifier boundaries: {'1': [(4, 2044, 1), (0, 60, 1)], '2': [(4, 2044, 1), (512, 60, 1)], '3': [(4, 2044, 1), (1024, 60, 1)], '4': [(4, 2044, 1), (1536, 60, 1)]}
08/25/2020 10:09:28 AM INFO: 	Creating stack of CDS difference frames
08/25/2020 10:09:28 AM INFO: 	Creating readnoise image
08/25/2020 10:09:28 AM INFO: 	Readnoise image saved to /grp/jwst/ins/jwql/outputs/readnoise_monitor/data/niriss_nis_wfss64c/jw87600023001_02105_00001_nis_uncal_refpix_readnoise.fits
08/25/2020 10:09:28 AM INFO: 	Readnoise image stats: 9.65369 +/- 2.65065
08/25/2020 10:09:28 AM WARNING: readnoise_monitor.py:190: RuntimeWarning: Mean of empty slice
  amp_stats['amp{}_mean'.format(key)] = np.nanmean(clipped)

[... cut ...]
    bin_edges, uniform_bins = _get_bin_edges(a, bins, range, weights)
  File "/grp/jwst/ins/jwql/envs/miniconda3/envs/jwql-3.6/lib/python3.6/site-packages/numpy/lib/histograms.py", line 387, in _get_bin_edges
    first_edge, last_edge = _get_outer_edges(a, range)
  File "/grp/jwst/ins/jwql/envs/miniconda3/envs/jwql-3.6/lib/python3.6/site-packages/numpy/lib/histograms.py", line 307, in _get_outer_edges
    "supplied range of [{}, {}] is not finite".format(first_edge, last_edge))
ValueError: supplied range of [nan, nan] is not finite
```

bsunnquist, 2020-08-25:
those Amplifier Boundaries look off - I think what happened is the WFSS128C and WFSS64C subarrays were added back into the `FOUR_AMP_SUBARRAYS` list in `constants.py`. After talking with Kevin, we decided these didn't belong in that list, so I removed them in the readnoise monitor pull request, but it looks like they've since been added back in.

bourque, 2020-08-25:
Thanks @bsunnquist!  I'll track that down and see where we went wrong.

bsunnquist, 2020-08-25:
ok, I think I'll add some tests for `instrument_properties.amplifier_info()` in the readnoise monitor tests to make sure the `amp_bounds` that get returned are reasonable

bourque, 2020-08-25:
That would be great!

bourque, 2020-08-26:
@bsunnquist I looked into the history of `constanst.py` and I am really perplexed how those got added back in there.  The commit after yours (https://github.com/spacetelescope/jwql/commit/2b5febcbd1fe3497821b5c4e18f698b81ae96271#diff-0417e3043d1a4c1857f9e22ae3dfa3fe) does have those subarrays in the list, but it is not shown as a change to the file (https://github.com/spacetelescope/jwql/commit/2b5febcbd1fe3497821b5c4e18f698b81ae96271#diff-0417e3043d1a4c1857f9e22ae3dfa3feL76) 🤔 Very strange!

In any case, I will make a small commit to `develop` that re-removes those, and then try to run the readnoise monitor again.

bourque, 2020-08-26:
Alright, that seemed to solve it!  Thanks for your help @bsunnquist.  Closing this issue.

closed 2020-08-26 by bourque