spacetelescope/jwql #627: Bad pixel monitor is failing in cron job

bhilbert4, 2020-09-04:
From the email sent out early this morning, there is an import error related to pandas.

```
Traceback (most recent call last):
  File "bad_pixel_monitor.py", line 88, in <module>
    from jwql.database.database_interface import session
  File "database_interface.py", line 64, in <module>
    import pandas as pd
  File "jwql/envs/miniconda3/envs/jwql-3.6/lib/python3.6/site-packages/pandas/__init__.py", line 22, in <module>
    from pandas.compat.numpy import (
ImportError: cannot import name '_np_version_under1p14'
```

bourque, 2020-09-08:
@bhilbert4 Thanks for opening an issue for this.  This one has really perplexed me.  It seems that this can be temporarily fixed by re-installing `pandas` in the conda environment, but I find that by the next day we get the same error again 🤷

bhilbert4, 2020-09-08:
Looks like the ~latest version of pandas (master branch from a month ago) has dropped support for python 3.6. Maybe that's part of the issue? 
https://github.com/pandas-dev/pandas/pull/35214

The minimum supported numpy version in the pandas master branch is 1.16.5, while our python 3.6 env is using 1.16.4. I know that our 3.6 env is using an older version of pandas than the pandas master branch from a month ago, but maybe we should try bumping up the numpy version?

bourque, 2020-09-09:
Awesome investigative work!  Yes, I will try tweaking some of these versions.  This also raises a new question -- should the 'default' JWQL version (the one that we deploy and use to run cron jobs and the web app on the servers) be updated to 3.7?  From what I can gather, more people are using 3.7 nowadays than 3.6.  Something to think about.

PR #629 (2020-09-09): Update pandas and numpy to solve cron job issues
This PR updates the versions of `pandas` and `numpy` in attempt to solve the issue presented in #627.

Closes #627
closed 2020-09-09 by bourque