spacetelescope/jwql #658: Demo the JWQL application to interested teams prior to 1.0 release

bourque, 2020-11-02:
When we are about ready for our 1.0 release, we should demo the application to interested parties to show the current capabilities and solicit useful feedback.

bourque, 2020-11-02:
Interested teams may include:

- NIRCam Instrument Team
- NIRISS Instrument Team
- NIRSpec Instrument Team
- MIRI Instrument Team
- TEL/FGS Instrument Teams
- DMS Team(s)
- ASB Team(s)
- INS Management Team

mfixstsci, 2021-04-20:
@bourque if you needed any assistance with this, I would be more than willing to split duties and present with you when the time comes.

bourque, 2021-04-21:
Thanks @mfixstsci!  Perhaps you could lead the web app demo portion of these presentations if you'd like.

mfixstsci, 2021-04-21:
@bourque I would be totally okay with that. I can also help out with the presentation as well. I've seen some of your talks in the past and it would be a good opportunity for me to brush up on my power point skills :sweat:

bourque, 2021-04-21:
@mfixstsci Sounds good.  Let's tag up about it and come up with a plan for slides after you return from vacation.

mfixstsci, 2021-04-21:
Cool, I assigned myself to the issue too so I don't forget.

bourque, 2021-05-25:
I will update this comment with the scheduled meetings as they come in:

- ASB et al: July 21 15:00

bourque, 2021-06-08:
All of the meetings have been scheduled!

mfixstsci, 2021-06-08:
Thanks @bourque looking forward to it!

mfixstsci, 2021-06-08:
Road Map
----------
ADD STEPS HERE (THESE CAN CHANGE)

* Step up presentations to be the same for each instrument for a focus on their interest.
* Meetings with management/mission office/DMS should be a high level overview of the app show casing more of the back end. 
* For DMS, don't focus too much on instrument monitors, more on EDB, preview images etc
* Meeting with ASB, focus more on authentication protocols (Mission Office too, mainly because of proprietary data restrictions).

mfixstsci, 2021-09-07:
> All of the meetings have been scheduled!

@bourque and they have all been done!

closed 2021-09-07 by mfixstsci