spacetelescope/jwql #667: Bad Pixel Monitor failing on dljwql

bhilbert4, 2020-11-19:
For some reason, the bad pixel monitor is failing on dljwql. When I run it locally, the page is rendered and everything looks fine. But on the server I see the traceback below. One difference: locally I'm running using python 3.8. 

```
KeyError at /nircam/bad_pixel_monitor/
'dead_history_yrange'

Request Method: GET
Request URL: http://localhost/nircam/bad_pixel_monitor/
Django Version: 2.1.7
Exception Type: KeyError
Exception Value: 'dead_history_yrange'
Exception Location: monitor_bad_pixel_bokeh.py in _update_badpix_v_time, line 373
Python Version: 3.6.4
```

bourque, 2020-11-19:
This issue might fix itself once we switch the dev server to use the `jwql-3.7` environment instead of the `jwql-3.6` environment, which is in the works (see #648)

bhilbert4, 2020-11-19:
I was wondering about that. I just logged in, and the bash profile starts up the 3.7 environment by default. Is that different than the environment that's being used to run jwql?

bourque, 2020-11-19:
Yeah, the web server is using a separate environment installation location, which is also something I am asking IT to change for us.  Once they do, the web app will be using the same 3.7 environment that you mention, which is also used to run the cron jobs.

bourque, 2021-02-03:
@bhilbert4 It looks like the Bad Pixel Monitor is now failing for a different reason:

```
FileNotFoundError: [Errno 2] No such file or directory: '/home/svc_jwqladm_dev/jwql/miniconda3/envs/jwql-3.7/lib/python3.7/site-packages/jwst_reffiles/bad_pixel_mask/miri_mirifulong_manual_flags.txt'
```

Could you address this as part of this Issue/sprint?

bhilbert4, 2021-02-03:
That's very strange. I see that file in my local copy of `jwst_reffiles`. Maybe there was some problem with the cloning process?

bhilbert4, 2021-02-11:
It looks like none of the `txt` files from the `bad_pixel_mask` directory in `jwst_reffiles` are present in the installation on dqjwql.  The 4 expected `.py` files are there. The low epoxy region definition files (which are not `py` files) are also missing from the utils directory. I believe this is a problem with the jwst_reffiles packaging such that data files are not being included.

bhilbert4, 2021-02-11:
jwst_reffiles was fixed in https://github.com/spacetelescope/jwst_reffiles/pull/125 
@bourque JWQL re-creates the environment on dljwql pretty often, right? Should we just let it do that and then test again?

bourque, 2021-02-25:
Thank you for this fix @bhilbert4!  Upon testing the bad pixel monitor out again, it completed successfully, so I think we can close this issue!

closed 2021-02-25 by bourque