spacetelescope/jwql #685: Switch master branch to 'main'

bourque, 2021-01-08:
The latest version of `git` call the default branch `main` now instead of `master`, so we should change our branch name to reflect that.

mfixstsci, 2021-03-10:
@bourque I can do this one, do I need special privileges to have access to to the settings tab of the repository to rename the default branch?

bourque, 2021-03-10:
@mfixstsci Good question, I'm not really sure, but I've given you Admin access for the time being in order to do this!

bourque, 2021-03-10:
@mfixstsci When you do this, could you also make sure that any references to `master` in our documentation are also switched over? I'm not sure there are any, but something to check.

mfixstsci, 2021-03-10:
@bourque can do!

bourque, 2021-03-18:
This is now completed.  Thanks @mfixstsci!

closed 2021-03-18 by bourque

mfixstsci, 2021-03-18:
Thanks @bourque should have closed the loop! Pushing directly to the `upstream` remote is weird.

bourque, 2021-03-18:
Haha yeah, it feels wrong ...