spacetelescope/jwql #704: Changes to recent versions of jwst causing pipeline_tools to crash

bourque, 2021-03-01:
It looks like recent versions of `jwst` (`<=0.18.3` or so) have changed the way `jwst.rscd` is used:

```
Traceback (most recent call last):
  File "cosmic_ray_monitor.py", line 51, in <module>
    from jwql.instrument_monitors import pipeline_tools
  File "/Users/bourque/Desktop/repositories/jwql/jwql/instrument_monitors/pipeline_tools.py", line 36, in <module>
    from jwst.rscd import RSCD_Step
ImportError: cannot import name 'RSCD_Step' from 'jwst.rscd' (/Users/bourque/miniconda3/envs/jwql-3.7/lib/python3.7/site-packages/jwst/rscd/__init__.py)
```

This causes `pipeline_tools` to crash.  It would be good if we could get this working so that JWQL can keep up with the latest versions of `jwst`

bourque, 2021-03-01:
@bhilbert4 @bsunnquist As our resident pipeline experts, I was wondering if one of you could take a look at this if you have some extra time?

bsunnquist, 2021-03-01:
I can take a look at it - looks like it might just be the RSCD call name

bsunnquist, 2021-03-01:
@bourque I put a fix in this PR (https://github.com/spacetelescope/jwql/pull/705) - though it will only work with the newer jwst pipeline versions, so I'm not sure if you'd want to merge it until our jwql environments use those newer pipeline versions.

closed 2021-03-02 by bourque