spacetelescope/jwql #735: Various improvements to the monitor status table in the dashboard view

bourque, 2021-05-17:
Upon testing #726, I found a few issues and thought of a few changes/improvements we could make to the monitor status table in the [dashboard](https://dljwql.stsci.edu/dashboard/):

- The `start_time` is not color coded like the rest of the table
- The `start_time` and `end_time` columns need not decimal precision in seconds.  A `YYYY-MM-DD HH:MM:SS` value would suffice.
- Some rows have `status` values of `None` (and thus row color of yellow) even though the corresponding log files indicate that the monitor completed successfully.  (Could this be an issue with `monitor_cron_jobs.py`?)
- It appears that the `jwql_monitors` (e.g. `monitor_filesystem`, `monitor_mast`, etc.) are not included in the table.  It would be good to include these.

bourque, 2021-09-13:
@mfixstsci Here is some more low-hanging fruit if you are looking for something to work on in the coming weeks!

mfixstsci, 2021-09-14:
Hey @bourque,

- The table library we use doesn't color coat the column selected to organize the table by. By default I selected the table to sort by the most recent run, which is why the date column is grey.
- I can change the precision on the dates no problem.

mfixstsci, 2021-09-20:
> Hey @bourque,
> 
>     * [ ]  The table library we use doesn't color coat the column selected to organize the table by. By default I selected the table to sort by the most recent run, which is why the date column is grey.
> 
>     * [x]  I can change the precision on the dates no problem.

I went ahead and submitted a PR #796. Should I keep digging on the column highlighting? Or are we okay with the column being grey?

bourque, 2021-09-22:
Yeah no worries about the column coloring if it is going to be a lot of work.  It's a minor thing.