spacetelescope/jwql #736: 20 May 2021: Trouble logging in to MAST in order to access EDB data

bhilbert4, 2021-05-20:
When on dljwql.stsci.edu, when I click on EDB and then query the EDB database, it asks me to log in via MAST. I do that and then return to the EDB page, but when I query the database again, it again asks me to log in via MAST. My username is on the top right of the page, so I believe I am successfully logged in.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to EDB
2. Click on the button to query the EDB with the default telemetry mnemonic
3. Log in via MAST
4. Repeat step 2

bourque, 2021-05-20:
Thank you @bhilbert4.  Likely something got messed up with the token stored on the dev server (again).  I will investigate and fix within the coming weeks.

bourque, 2021-06-17:
This should be fixed now.  For some reason, the MAST token that we were using was removed from `auth.mast`.  Perhaps tokens are removed after some time period automatically?  In any case, I created a new token and put that into the `config.json` file, and things are working now.

closed 2021-06-17 by bourque