spacetelescope/jwql #739: 8 June 2021: Enable readthedocs test build with PRs

bhilbert4, 2021-06-08:
#724 passed all CI tests before being merged, but a check of readthedocs shows that the build there is now failing.

We should enable a test readthedocs build with each PR so that we can more easily catch documentation errors. I've been using this with Mirage for over a year now, and it's very helpful. Instructions for setting this up are here (looks much easier to set up then it was when I enabled it for Mirage): https://docs.readthedocs.io/en/stable/pull-requests.html

I don't see the "Admin" tab they mention, so I must not have the permissions to enable this.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

bhilbert4, 2021-06-08:
Log of the failing build: 
https://readthedocs.org/projects/jwql/builds/13958097/

bourque, 2021-06-08:
Thanks @bhilbert4, this is a great idea!

bourque, 2021-06-17:
This is now set up, however it appears that the builds get stuck on "pending" (e.g. https://github.com/spacetelescope/jwql/pull/726) even though when clicking on "Details", the build is successful on RTD.

bourque, 2021-06-17:
Nevermind, the stuck RTD build appears to have fixed itself 🤷

closed 2021-06-17 by bourque