spacetelescope/jwql #751: [DATE]: [Bug]

bhilbert4, 2021-07-06:
The latest cron job emails indicate that our disk space is full on the dev server:

-- Logging error ---
Traceback (most recent call last):
  File "jwql/miniconda3/envs/jwql-3.7/lib/python3.7/logging/__init__.py", line 1029, in emit
    self.flush()
  File "jwql/miniconda3/envs/jwql-3.7/lib/python3.7/logging/__init__.py", line 1009, in flush
    self.stream.flush()
OSError: [Errno 28] No space left on device

bourque, 2021-07-06:
Thanks @bhilbert4.  This is likely due to `generate_preview_images` creating a whole bunch of new preview images from the latest DMS build area (#748).  I will investigate.

bourque, 2021-07-14:
This should be fixed now.  The preview images from #748 were taking up all of the internal storage.  I requested and ITSD supplied another 1TB of storage.

closed 2021-07-14 by bourque