spacetelescope/jwql #762: Readnoise Monitor crashes on data from latest DMS build

bourque, 2021-07-19:
**Describe the bug**
The `readnoise_monitor` crashes on MIRI data from the latest DMS build (#748)

**To Reproduce**
Steps to reproduce the behavior:
Run the `readnoise_monitor.py` script on the dev server which points to the latest DMS build

Here is some of the log file that contains the traceback (some info has been clipped for redundancy and not exposing directory structures)

```
07/19/2021 11:42:09 AM INFO: 	Working on file: outputs/dev/readnoise_monitor/data/miri_mirim_brightsky/jw00873005001_02101_00001_mirifushort_uncal.fits
07/19/2021 11:42:09 AM INFO: 	Running pipeline on /outputs/dev/readnoise_monitor/data/miri_mirim_brightsky/jw00873005001_02101_00001_mirifushort_uncal.fits
07/19/2021 11:42:11 AM INFO: DQInitStep instance created.
07/19/2021 11:42:11 AM INFO: Step DQInitStep running with args ('outputs/dev/readnoise_monitor/data/miri_mirim_brightsky/jw00873005001_02101_00001_mirifushort_uncal.fits',).
07/19/2021 11:42:11 AM INFO: Step DQInitStep parameters are: {'pre_hooks': [], 'post_hooks': [], 'output_file': None, 'output_dir': None, 'output_ext': '.fits', 'output_use_model': False, 'output_use_index': True, 'save_results': False, 'skip': False, 'suffix': None, 'search_output_file': True, 'input_dir': ''}
07/19/2021 11:42:11 AM DEBUG: Input model type Level1bModel does not match the requested model <class 'jwst.datamodels.ramp.RampModel'>. Cannot skip updating from FITS headers.
07/19/2021 11:42:12 AM INFO: Using MASK reference file references/jwst/jwst_miri_mask_0022.fits
07/19/2021 11:42:12 AM DEBUG: No ASDF information found. Cannot skip updating from FITS headers.
07/19/2021 11:42:12 AM DEBUG: ref substrt1=1, subsize1=1032, substrt2=1, subsize2=1024
07/19/2021 11:42:12 AM DEBUG: sci substrt1=1, subsize1=1032, substrt2=1, subsize2=1024
07/19/2021 11:42:12 AM INFO: Step DQInitStep done
07/19/2021 11:42:12 AM INFO: RefPixStep instance created.
07/19/2021 11:42:12 AM INFO: Step RefPixStep running with args (<RampModel(1, 16, 1024, 1032) from jw00873005001_02101_00001_mirifushort_uncal.fits>,).
07/19/2021 11:42:12 AM INFO: Step RefPixStep parameters are: {'pre_hooks': [], 'post_hooks': [], 'output_file': None, 'output_dir': None, 'output_ext': '.fits', 'output_use_model': False, 'output_use_index': True, 'save_results': False, 'skip': False, 'suffix': None, 'search_output_file': True, 'input_dir': '', 'odd_even_columns': True, 'use_side_ref_pixels': True, 'side_smoothing_length': 11, 'side_gain': 1.0, 'odd_even_rows': True}
07/19/2021 11:42:12 AM INFO: use_side_ref_pixels = True
07/19/2021 11:42:12 AM INFO: odd_even_columns = True
07/19/2021 11:42:12 AM INFO: side_smoothing_length = 11
07/19/2021 11:42:12 AM INFO: side_gain = 1.000000
07/19/2021 11:42:12 AM INFO: odd_even_rows = True
07/19/2021 11:42:12 AM INFO: Subtracting initial read from each integration
07/19/2021 11:42:12 AM INFO: Adding initial read back in
07/19/2021 11:42:12 AM INFO: Step RefPixStep done
07/19/2021 11:42:13 AM INFO: 	Pipeline complete. Output: outputs/dev/readnoise_monitor/data/miri_mirim_brightsky/jw00873005001_02101_00001_mirifushort_uncal_refpix.fits
07/19/2021 11:42:13 AM INFO: 	Amplifier boundaries: {'1': [(4, 1028, 4), (0, 1024, 1)], '2': [(5, 1028, 4), (0, 1024, 1)], '3': [(6, 1028, 4), (0, 1024, 1)], '4': [(7, 1028, 4), (0, 1024, 1)]}
07/19/2021 11:42:13 AM INFO: 	Creating stack of CDS difference frames
07/19/2021 11:42:13 AM INFO: 	Creating readnoise image
07/19/2021 11:42:15 AM INFO: 	Readnoise image saved to /outputs/dev/readnoise_monitor/data/miri_mirim_brightsky/jw00873005001_02101_00001_mirifushort_uncal_refpix_readnoise.fits
07/19/2021 11:42:15 AM INFO: 	Readnoise image stats: 13399.20041 +/- 13315.16238
07/19/2021 11:42:15 AM INFO: 	Readnoise image stats by amp: {'amp1_mean': 13401.957523689449, 'amp1_stddev': 13312.881117000146, 'amp1_n': array([57420,     0,     0,     0, 98210,     0,     0,     0,     0,
           0,     0,     0,     0,     0,     0,     0,     0,     0,
           0,     0,     0,     0,     0,     0,     0,     0,     0,
           0,     0,     0,     0,     0,     0,     0,     0,     0,
       40644, 16232,     0,     0, 16234, 25226,     0,     0,     0,
        8178,     0,     0,     0,     0,     0,     0,     0,     0,
           0,     0,     0,     0,     0,     0,     0,     0,     0,
           0,     0,     0,     0,     0,     0,     0,     0,     0,
           0,     0,     0,     0,     0,     0,     0,     0,     0,
           0,     0,     0,     0,     0,     0,     0,     0]), 'amp1_bin_centers': array([  374.457764  ,  1123.37329199,  1872.28881999,  2621.20434799,
        3370.11987598,  4119.03540398,  4867.95093198,  5616.86645997,
        6365.78198797,  7114.69751597,  7863.61304396,  8612.52857196,
        9361.44409996, 10110.35962795, 10859.27515595, 11608.19068395,
       12357.10621194, 13106.02173994, 13854.93726794, 14603.85279593,
       15352.76832393, 16101.68385193, 16850.59937992, 17599.51490792,
       18348.43043591, 19097.34596391, 19846.26149191, 20595.1770199 ,
       21344.0925479 , 22093.0080759 , 22841.92360389, 23590.83913189,
       24339.75465989, 25088.67018788, 25837.58571588, 26586.50124388,
       27335.41677187, 28084.33229987, 28833.24782787, 29582.16335586,
       30331.07888386, 31079.99441186, 31828.90993985, 32577.82546785,
       33326.74099585, 34075.65652384, 34824.57205184, 35573.48757983,
       36322.40310783, 37071.31863583, 37820.23416382, 38569.14969182,
       39318.06521982, 40066.98074781, 40815.89627581, 41564.81180381,
       42313.7273318 , 43062.6428598 , 43811.5583878 , 44560.47391579,
       45309.38944379, 46058.30497179, 46807.22049978, 47556.13602778,
       48305.05155578, 49053.96708377, 49802.88261177, 50551.79813976,
       51300.71366776, 52049.62919576, 52798.54472375, 53547.46025175,
       54296.37577975, 55045.29130774, 55794.20683574, 56543.12236374,
       57292.03789173, 58040.95341973, 58789.86894773, 59538.78447572,
       60287.70000372, 61036.61553172, 61785.53105971, 62534.44658771,
       63283.36211571, 64032.2776437 , 64781.1931717 , 65530.1086997 ,
       66279.02422769]), 'amp2_mean': 13398.180153590465, 'amp2_stddev': 13315.952386198795, 'amp2_n': array([57421,     0,     0,     0, 98210,     0,     0,     0,     0,
[... 87 lines cut ...]
       51307.26050922, 52056.27161155, 52805.28271387, 53554.2938162 ,
       54303.30491852, 55052.31602085, 55801.32712317, 56550.33822549,
       57299.34932782, 58048.36043014, 58797.37153247, 59546.38263479,
       60295.39373712, 61044.40483944, 61793.41594177, 62542.42704409,
       63291.43814641, 64040.44924874, 64789.46035106, 65538.47145339,
       66287.48255571])}
07/19/2021 11:42:15 AM CRITICAL: Traceback (most recent call last):
  File "wql/jwql/jwql/utils/logging_functions.py", line 284, in wrapped
    func(*args, **kwargs)
  File "jwql/jwql/jwql/utils/logging_functions.py", line 249, in wrapped
    func(*args, **kwargs)
  File "readnoise_monitor.py", line 593, in run
    self.process(new_files)
  File "readnoise_monitor.py", line 447, in process
    reffile_mapping = crds.getreferences(parameters, reftypes=['readnoise'])
  File "jwql-3.7/lib/python3.7/site-packages/crds/core/heavy_client.py", line 128, in getreferences
    final_context, bestrefs, ignore_cache=ignore_cache)
  File "jwql-3.7/lib/python3.7/site-packages/crds/client/api.py", line 848, in cache_references
    wanted = _get_cache_filelist_and_report_errors(bestrefs)
  File "jwql-3.7/lib/python3.7/site-packages/crds/client/api.py", line 891, in _get_cache_filelist_and_report_errors
    raise last_error
crds.core.exceptions.CrdsLookupError: Error determining best reference for 'readnoise'  =   No match found.

07/19/2021 11:42:15 AM CRITICAL: CRASHED
```

bsunnquist, 2021-07-23:
I can put in a fix for this. It looks like it isn't robust to missing reffiles like I thought it was.

My hunch is that we're now picking up rehearsal test data (GSEG, LRE, etc), which can include strange detector setups and test pattern data (which would also explain why the results for some of these images are so strange).

bourque, 2021-07-23:
Thank you @bsunnquist!

Do you think the problem that you describe would exist when we are in operations?  Would it be easier to just not process rehearsal test data?

bsunnquist, 2021-07-26:
Either way is fine on my end - if we still process it, I may just eventually cut the start range on my monitor plots to exclude any pre-launch data, so these strange test pattern results don't confuse people.

The test pattern data won't be an issue in commissioning/operations. Here is an example of it, basically the pixels values just increment from 1 to 65535 over and over, so any analysis/pipeline results are always strange looking.
[image]

I made the Readnoise Monitor robust to the other issue as well (missing reffiles) here https://github.com/spacetelescope/jwql/pull/766, so it won't crash it that scenario anymore. This should also be rare/non-existent in commissioning (could just be an issue if teams forget to make reffiles for certain modes).

bourque, 2021-07-26:
Perfect, sounds great! Thanks

closed 2021-07-27 by bourque