spacetelescope/jwql #769: Ask instrument teams for updated priorities for monitoring

bourque, 2021-07-29:
After our 1.0 release, we will be able to start developing new monitors for the application.  I think ~2018 was the last time we asked teams about which monitors they would like and what the priorities were.  We should ask the instrument teams about this and update our priorities if necessary.

bourque, 2021-10-07:
@bhilbert4 Yes, that was my intention

bsunnquist, 2021-10-07:
Looking at our instrument monitor overview page, I think the Pedestal Monitor is already taken care of in the Bias Monitor (it tracks the average 0th group levels over time in darks) - it's marked 'critical' for NIRCam/NIRISS, so we could change those to complete if we're updating that page as part of this task

bhilbert4, 2021-10-08:
For NIRCam, a discussion on slack has come up with several monitors that folks think should be high priority. Roughly in order of highest to lowest priority:

Data trending
TA monitor (look at repeatability/uncertainty of small angle maneuvers)
PSF characterization (encircled energy, etc)
Photometric Stability Monitor
Sky/telescope Background Monitor

bourque, 2021-11-01:
Just noting here that @cracraft has indicated that MIRI is most interested in a TA monitor and Sky/Background Monitor.

Now that we have updated priorities from each instrument team, I think we can close this issue.

closed 2021-11-01 by bourque