spacetelescope/jwql #771: 30 July 2021: Reset monitor databases at start of commissioning

bhilbert4, 2021-07-30:
**Describe the bug**
Now that we have switched to using the isilon filesystem, the instrument monitors are working on real data generated during rehearsals. While these data have realistic headers, the pixel values in the data are garbage. This means that the dark rates, readnoise and bias levels, and bad pixel lists generated using these data are also garbage. Assuming we have all the monitors working well just prior to commissioning, we should reset all of the monitor database tables and limit the MAST queries used by the monitors to start no earlier than commissioning. This will ensure that rehearsal data will not be used from that point onward. While it will also preclude the use of real ground testing data, it's probably safe to assume that all good ground testing data will have been turned into reference files and delivered to CRDS by that point. Most monitors check new results against existing reference files, so ground testing results will still be used.

bhilbert4, 2021-08-03:
Another option we could try would be to keep results from ground testing data, but ignoring rehearsal data. This would have the advantage of showing accurate monitor results for the ground testing campaigns, which could be helpful, especially during commissioning and cycle 1 calibration. In order to accomplish this we'd have to find a way to ignore data taken between ~2017 and the end of 2021.

bsunnquist, 2021-08-24:
Would the ground testing data (e.g. CV3/OTIS) get picked up by the mast query function, or is it not available in mast? 

Those would be useful to include for the readnoise/bias pages, but perhaps not critical, since like you said these monitors compare to ground reffiles anyway.

bourque, 2022-01-31:
It is now time to do this!  @bhilbert4 @bsunnquist Do you have any new-found opinions on how we should handle the MAST queries / query times in the monitors?  Once we point the monitors to query the OPS version of `astroquery.mast`, there shouldn't be any rehearsal data picked up by it anyways right?  Since the DMS folks cleared out all of the rehearsal data prior to commissioning?

bhilbert4, 2022-01-31:
I haven't played with MAST at all to see what's left in there. Is it essentially empty now? Did they remove all ground testing data? If so then I guess we can just set the minimum/initial query date to ~launch?

bourque, 2022-02-14:
I can confirm that the production side of `astroquery.mast` is returning expected results, and is not returning any rehearsal data.

The databases on dev, test, and ops have been reset, so this ticket can now be closed.

closed 2022-02-14 by bourque