spacetelescope/jwql #775: 2 Aug 2021: Readnoise monitor failing with failure to find readnoise reference file

bhilbert4, 2021-08-03:
**Describe the bug**
It looks like the readnoise monitor is failing on the dev server when crds is unable to find a readnoise reference file when working on the MIRIM_BRIGHTSKY aperture. See the log file: readnoise_monitor_2021-08-02-04-30.log 

@bsunnquist

bsunnquist, 2021-08-04:
Looks like the code hasn't picked up the new changes yet from https://github.com/spacetelescope/jwql/pull/766 - but once it does this should be fixed. 

@bourque should this version be picked up already though, since it was merged about a week ago into the dev branch?

bsunnquist, 2021-08-24:
Just noting, it looks like this code is working again as expected based on the August 24 log file - those new changes must have been applied.

bourque, 2021-08-24:
Thanks @bsunnquist, I think you're right.  As such, I am going to close this issue.

closed 2021-08-24 by bourque