spacetelescope/jwql #778: EDB mnemonics plotting results does not handle certain data types

bourque, 2021-08-03:
Currently the EDB mnemonics plotting feature works on time-series data (for example, the `IMIR_HK_ICE_SEC_VOLT4` mnemonic), but does not gracefully handle other data types such as string values (for example, the `IFGS_CTD_GS` mnemonic).

We should either find a way to support other data types (e.g. return a table of values instead of a plot?), or put in an error message stating that the mnemonic is of an unsupported datatype.

mfixstsci, 2021-09-02:
I will take this one!

bourque, 2021-12-16:
@mfixstsci I'm just checking back in on this one.  IIRC, you implemented a warning to the user if data is not able to be returned?  If that is the case, I'm inclined to close this issue.

bourque, 2022-01-26:
I tried out the EDB for a mnemonic that returns string data (`IFGS_CTD_GS`) and see there is a message: "QUERY RESULT RETURNED NO DATA FOR IFGS_CTD_GS ON DATES 2019-04-02 00:00:00.000 - 2019-04-02 00:01:00.000", so I think this issue can be closed.

closed 2022-01-26 by bourque

mfixstsci, 2022-01-26:
@bourque thanks for closing, sorry I didn't respond earlier 😓 I remember looking into the non-numeric mnemonic issue when I restructured the EBD interface. Do we have a list of mnemonics with non-numerical data? I don't mind taking a deeper dive on this if needed.