spacetelescope/jwql #779: Add more information to the About page

bourque, 2021-08-03:
Currently the About page on the web app describes an overview of the project, the technologies used, an overview of the instrument monitoring, and a description of the development team.

There are number of things we could add to this page to more thoroughly describe the project and how users may interact with it.

Here is a running list of things to add (or point to existing documentation via a hyperlink):
- [ ] Describe the process for submitting a bug report and/or feature request
- [ ] Describe the process for authentication
- [ ] Describe the process for contributing software changes to the project
- [ ] Describe which data the application serves and where it comes from
- [ ] Describe which monitors we support and where the algorithms come from
- [ ] Add a table of contents

bhilbert4, 2021-08-03:
We'll have to draw a line somewhere that separates information to go on the About page versus information to go on Readthedocs, or else we risk ending up with a huge About page that makes it hard to find information.

bourque, 2021-08-25:
We should take out the `Describe the process for submitting a bug report and/or feature request` item and make it part of the v1.0 release.

PR #853 (2022-01-14): Update About page
Closes #779
closed 2022-01-14 by bourque