spacetelescope/jwql #780: Determine final list of users for AD group

bourque, 2021-08-03:
The JWQL web app will authenticate against an Active Directory (AD) group.  We need to define this AD group to determine who will ultimately have access to the website in production.  We can do this by asking team leads to identify individuals who should have access to the app.

bhilbert4, 2021-09-29:
It looks like @marthaboyer may be the NIRCam person to ping about this. What kind of access will people not in this group have? At the moment, it looks like you don't need to log in to see the results from the dark/bias/readnoise/etc monitors? Will it be the same way once everything is behind the SSO, or will sign in be necessary for all parts of the website?

bourque, 2021-09-29:
Thanks @bhilbert4.  Only individuals in the AD group will be able to access the web app.  That is to say, all parts of the web app (including monitoring results) will be behind the SSO/AD group firewall.

marthaboyer, 2021-09-29:
Sorry, I thought Massimo had sent you the list already. I'll send you an email.

bourque, 2021-09-29:
Thank you @marthaboyer!

This issue is now complete!

closed 2021-09-29 by bourque