spacetelescope/jwql #783: Investigate dashboard monitor results

bourque, 2021-08-03:
Many of the monitor runs in the monitor table in the Dashboard view of the web app reports that most of the monitors are crashing (red) or have not completed successfully but have not crashed (yellow).  We should investigate what is going on here.  Are the monitors really not completing all the time? Or is there a possible bug somewhere?

mfixstsci, 2021-08-24:
Interesting, it looks like there are three type of status result from the monitor table.

```
postgres_databse=> SELECT DISTINCT status FROM monitor;
 status  
---------
 SUCCESS
 
 FAILURE
```
`SUCCESS " " and FAILURE` But, when we count
```
postgres_database=> SELECT COUNT(DISTINCT status) FROM monitor;
 count 
-------
     2
(1 row)
```
we only get 2 entries. I am guessing because we don't count `None` types in the database tables. This must be set when the monitors are run.

mfixstsci, 2021-08-24:
Looks like it's a bug :(

```
import os
from jwql.jwql_monitors.monitor_cron_jobs import success_check

# Entries in the database for these logs
# [NONE, NONE, SUCCESS, FAILURE]
logs = ['bias_monitor_2020-11-28-03-30.log', 'readnoise_monitor_2020-11-26-04-00.log', 'dark_monitor_2020-12-27-03-00.log', 'bad_pixel_monitor_2020-11-24-04-30.log']

path = '/path/to/jwql_logs/'

for log in logs:
    success_check(os.path.join(path, log))
```
and the results are:
```
success
success
success
failure
```

The logs for the `readnoise` and `bias` runs are successful but somewhere along the way, that information is lost.

bourque, 2021-08-29:
Thank you for finding that @mfixstsci!  I take fault for this one; redundant code like this should have never made it through the development and review process, sorry for sending you down a large rabbit hole! 🐇

closed 2021-09-02 by bourque