spacetelescope/jwql #835: Images not displaying in Readnoise Monitor

bsunnquist, 2021-12-02:
The png images aren't displaying on the NIRCam Readnoise (and probably the Bias) Monitor pages. This might be due to me storing their dev directory filepaths in the database tables whereas the web app looks in the ops directory.

bsunnquist, 2021-12-02:
i.e. see line ~133 in monitor_readnoise_bokeh.py
`diff_image_png = os.path.join('/static', '/'.join(diff_image_png.split('/')[-6:]))`

bsunnquist, 2021-12-03:
Update - I don't think this is a problem with the image path in the database for these monitors. They are correctly set to the ops path (not dev) and they exist.

bsunnquist, 2022-02-15:
The first flight data came in and these images are now displaying correctly! I'm not sure what changed, but just FYI @bourque I'm going to close this one.

closed 2022-02-15 by bsunnquist

bourque, 2022-02-15:
Thanks @bsunnquist!