spacetelescope/jwql #866: Oversized NIRCam display images

bsunnquist, 2022-02-03:
Some NIRCam preview images are oversized and overlap the anomaly checklist, e.g.:
https://jwql-dev.stsci.edu/NIRCam/jw01057005001_02102_00001_nrca1/
https://jwql-dev.stsci.edu/NIRCam/jw01057006001_02102_00001_nrcb1/

Others are fine though, e.g.:
https://jwql-dev.stsci.edu/NIRCam/jw01057003001_03101_00024_nrcblong/

bhilbert4, 2022-02-05:
The tick labels on the colorbars are also strange. In the SUBSTRIPE256 case, there are two sets of labels.

bhilbert4, 2022-02-05:
The code is set up such that all preview images have a length of 8 inches in their longest dimension. So it seems like this must be an issue of how the images are displayed, rather than the actual images themselves?

bhilbert4, 2022-02-07:
Sounds good. I also looked into the generate_preview_images code the other day and am pretty convinced that it's not explicitly making images that are too large in some cases but not others. I tried looking into the code that places the images in the html files to see if there was some issue with the display size, but I wasn't able to make sense of it enough to tell for sure.

BradleySappington, 2022-11-01:
@bsunnquist - is this still an issue you are encountering?

bsunnquist, 2022-11-01:
@BradleySappington Nope - it all looks good now! I'll go ahead and close this issue.

closed 2022-11-01 by bsunnquist