spacetelescope/jwql #890: Concise description of the feature request

cracraft, 2022-02-22:
Add scale to color bar in monitors with bokeh image displays.

On monitor pages like the dark monitor, in the Mean Dark Image tab, the color bar needs a scale to show what the colors actually represent. The image is yellow, but we don't know what yellow means.

Also, the x and y axes are not clear. If they are meant to be rows and columns of an image, a MIRI image is about 1024 by 1024 (1032 with reference pixels). The scale in the image on this page looks like it's a 10x10 pixel region, which doesn't make sense. This should be a bit clearer.

mfixstsci, 2022-03-01:
@bourque and @cracraft I found out the yellow square issue. When the DB is empty and there is no data available, a 10x10 array of zeros is generated to take the actual data's place. When the data is uniform, I think bokeh might have a hard time interpreting the values on the color bar.

As a test, I went and changed the from `data = np.zeros(10, 10)` to `data = np.random.rand(10, 10) + 1e3` and obtained this image:

[image]

Looks like bokeh templating works well, it is just a data issue. Maybe we can put in the title (empty database table, this is fake data) or something?

bourque, 2022-03-09:
It would be ideal if the webpage could just display some sort of "no data available" message instead of rendering a plot altogether, but if that is too complicated/difficult to implement, I think @mfixstsci's solution is a good option.

bhilbert4, 2024-05-13:
I believe this has been fixed.

closed 2024-05-13 by bhilbert4