spacetelescope/jwql #897: Missing NIRCam preview images due to mosaics

bhilbert4, 2022-03-04:
Currently, the web app shows many missing preview images for NIRCam data. This is because the preview image generator currently creates its own quick mosaics of preview images of NIRCam SW exposures. It creates these mosaic preview images, and skips the creation of preview images for individual detectors. However, in the web app pages for individual exposures, we have a separate page for each detector, and each of those pages is looking for a detector-specific preview image, rather than the mosaic.

We had originally thought that having these mosaic images would be a nice way to look at an entire exposure at once. However, with a separate page for each detector, combined with the fact that it will be hard to see small features/sources in a mosaic preview image, perhaps it makes more sense to go back to having detector-specific preview images for everything.

Maybe down the road we could add a link or a page that will show a full mosaic preview image if desired. But I'm thinking for the purposes of actual quick look, detector-specific might be better. 

@bourque @bsunnquist what do you think?

bsunnquist, 2022-03-04:
Yeah that definitely makes sense to me

bourque, 2022-03-09:
Yes, I agree as well!

PR #898 (2022-04-01): Strip out code for mosaic preview images
Getting a head start on this since I've been spending a lot of time today looking at the generate_preview_images.py already.

This PR removes the code that creates mosaic preview images for NIRCam observations, since the web app currently is not looking for these, and instead has separate pages for detector-specific preview images.

It might be best not to merge this until #867 is merged since they both have made changes to the same functions and there will probably be conflicts.

Resolves #897
closed 2022-04-01 by mfixstsci