spacetelescope/jwql #900: Readnoise Monitor not picking up new data

bsunnquist, 2022-03-11:
The Readnoise Monitor isn't running on new NIRCam darks, though the Bias Monitor is and they use very similar methods.

bsunnquist, 2022-03-21:
Update - it appear to be crashing due to a header keyword:
Failed validating 'enum' in schema:
    OrderedDict([('title', 'Quality of pointing information from EngDB'),
                 ('type', 'string'),
                 ('enum',
                  ['CALCULATED',
                   'CALCULATED_COARSE_TR_202107',
                   'CALCULATED_FULL',
                   'CALCULATED_FULLVA',
                   'CALCULATED_GSCMD',
                   'CALCULATED_GSCMD_V3PAGS',
                   'CALCULATED_ORIGINAL',
                   'PLANNED',
                   'CALCULATED_TR_202105',
                   'CALCULATED_TR_202105_VA',
                   'CALCULATED_TRACK_TR_202107']),
                 ('fits_keyword', 'ENGQLPTG'),
                 ('blend_table', True)])

On instance:
    'CALCULATED_COARSE_TR_202111'

I'm unable to reproduce the error locally and the pipeline runs fine (using and jwst 0.17.1), so this must be something new in the production jwst version 1.3.3

bsunnquist, 2022-03-21:
Yep that was it - thanks! I was able to reproduce the error on jwst1.3.3, but then it went away on jwst1.4.3.

Do you know how to get our ops environment to use jwst1.4.3 (i.e. the one jwql.stsci.edu uses)?

bhilbert4, 2022-03-21:
Ha. The pipeline, for versions >1.3.3, is supported only on python 3.8 and higher. So we need a whole new environment on the JWQL servers.

bsunnquist, 2022-03-21:
That's too bad. I do have a temporary workaround if needed - if I set `model.meta.visit.engdb_pointing_quality = 'CALCULATED'` before saving the model it runs fine (for some reason just removing that ENGQLPTG keyword from the header before running the pipeline doesn't work)

mengesser, 2022-03-22:
I don't see it noted anywhere so I thought I might add that the Dark monitor has also been failing on the same timeline for what I presume is the same reason.

bsunnquist, 2022-03-22:
Thanks @mengesser ! Yep, both the dark monitor and bad pixel monitor are also crashing due to this same issue (confirmed via their recent logs).

@bhilbert4 @mfixstsci I'm guessing the upgrade to 1.4.3 will be quite involved based on https://github.com/spacetelescope/jwql/issues/903 , so if you want I could just implement my quick workaround noted above in the meantime.

bsunnquist, 2022-03-25:
Per https://github.com/spacetelescope/jwql/issues/905, this should be fixed after I make the readnoise calculation in chunks to avoid memory issues

bsunnquist, 2022-04-07:
Update - I'm starting to suspect the memory issue is not the readnoise calculation itself. I was just able to run a 5x200x2048x2048 image through the readnoise calculation with no issues on our dev server.

bhilbert4, 2022-04-07:
Wow. That's interesting. I know when I compared older logs it was clear that multiple monitors were running at the same time, and I think the combined memory usage was a problem. Although if 1000 full frames work then I start to wonder about that conclusion.

bhilbert4, 2024-05-13:
This issue has been resolved.