spacetelescope/jwql #901: Possibility to select ONE Instrument when multiple Instruments are associated to a single Program (PID)

anoriegacrespo, 2022-03-11:
**Is your feature request related to a problem? Please describe.**
Right now multiple instruments e.g. OTE, NIRCAM & FGS are linked to a single identification program

**Describe the solution you'd like**
Given the PID, it would be good to be able to select ONE of the instrument, e.g NIRCAM

**Describe alternatives you've considered**
The above feature could be selected as a multiple choice, or specified as a Keyword, e.g. 01059 NIRCAM
or 01059 FGS

bhilbert4, 2022-03-23:
In terms of the back end, it looks like the search will need to be modified here:
https://github.com/spacetelescope/jwql/blob/develop/jwql/website/apps/jwql/forms.py#L275

bhilbert4, 2022-06-23:
Coming back around to this update. I was just testing an update to the filename parser and was playing with the form submission box. It looks to me like all programs are going to be considered multi-instrument simply because the guider data that goes along with each program is included. So for example, program 1442 is all NIRCam, but with the addition of the guide star data, the program becomes multi-instrument and the search returns a "I can't load a proposal with more than one instrument" warning. It might be nice to have the instrument selector at least default to the prime instrument for a given proposal so that users won't have to continually select their instrument in order to avoid guide star data.

mengesser, 2022-07-06:
I guess the question for this issue now is what exactly we want the query to look like. Does it make sense to type a PID, and then create a new field for selecting from the available instruments? Or does it make more sense to allow the user to give the PID and instrument together, and then throw an error if that combination isn't possible? What is possible with django forms?

bhilbert4, 2022-07-07:
I think the former makes more sense. @mfixstsci what do you think?

mengesser, 2022-07-26:
We have a solution for this that Mees developed. I will put it in a PR this week (Right now I am on a loaner waiting for an OS upgrade)

mfixstsci, 2022-08-23:
Addressed in #1010 changes should be in. @anoriegacrespo  if you have any questions, feel free to ask us. If the program has multiple instrument, links will appear to select the route you are interested in. If your program only has one instrument, you will be routed directly.

closed 2022-08-23 by mfixstsci