spacetelescope/jwql #904: (Concise description of bug)

bhilbert4, 2022-03-23:
While testing the readnoise monitor this morning, I saw a warning that will become an error down the road:

03/23/2022 12:35:43 PM INFO:    Creating png of readnoise difference image
03/23/2022 12:35:43 PM WARNING: savefig() got unexpected keyword argument "overwrite" which is no longer supported as of 3.3 and will become an error two minor releases later

I've searched matplotlib for details, but haven't found anything yet hinting at whether there is another keyword that replaces overwrite.

Once we find a solution for this, we should make sure that `overwrite` is removed in all JWQL calls to plt.savefig()

bsunnquist, 2022-04-21:
I also can't find an alternative - I've opened a matplotlib issue https://github.com/matplotlib/matplotlib/issues/22872

bsunnquist, 2022-04-21:
It turns out this keyword isn't even needed, the default is to overwrite anyway, fixed here https://github.com/spacetelescope/jwql/pull/937

closed 2022-04-22 by mfixstsci