spacetelescope/jwql #905: Failing monitors - probably due to memory issues with large files

bhilbert4, 2022-03-24:
It looks like all instrument monitors except the bias monitor are failing. In most cases, the logs from the monitor runs are not helpful. They show that the cal pipeline or some other calculation was starting, and that's the end of the log, implying that they are failing silently. 

For the bad pixel monitor, there is a line in the log saying that there was not enough memory to create an array for the data in the file being worked on. (It needed 1.5 GB). I looked in the logs for the other monitors and in all cases, the monitors were working on especially large files at the time that they failed. e.g. 4 integrations x 50 groups, 20 integrations x 10 groups. 

We may need to adjust the instrument monitors to be able to handle these cases. One option would be for the monitors to skip these files entirely. Or perhaps we can create a module that checks the number of groups and integrations and if a file is too large, read in and work on only a portion of the data? Unfortunately for the dark monitor, the failure came while the ramp-fitting step of the pipeline was running. I'm not sure how to get around that other than skipping the file entirely. We should check and see how large the official commissioning and cycle 1 cal dark data will be, in order to know whether this is going to continue to be a problem.

@bsunnquist

bhilbert4, 2022-03-24:
NIRCam's commissioning darks, at least the initial 3 visits for each module for the full frame apertures, are larger than the files mentioned above, so I would expect those to fail. The other 7 visits for each module are debatable. The files are smaller, but they are still quite large (1 int x 130 groups).

bsunnquist, 2022-03-25:
Yeah the commisioning data is huge. I'll check out the Readnoise Monitor - the Bias Monitor probably isn't failing because I have it extracting the 0th group only before running the pipeline.

The readnoise calculation is pretty memory intensive, but it's easy enough for me to do it in chunks if that's the issue. If it's the pipeline, I could probably just use the mast dark.fits files instead.

bhilbert4, 2022-03-25:
I tried spreading out the monitor runs last night in order to reduce overlap. Looks like things still crashed though. The readnoise monitor failed because it couldn't allocate 2.5 GB for an array. The final entry in the dark monitor log is at the same time as well, so it looks like that failed silently. 

We're going to ask IT for more memory, but I think anything we can do to reduce memory usage would help. I'll have to look at the dark monitor and see what I can do. I think the biggest memory hog there is just running the ramp-fitting step of the pipeline.

bsunnquist, 2022-03-25:
Ah ok, I'll switch to doing the readnoise calculation in chunks. Now I'm wondering if memory pressure might be one of the reasons for the overall website slowness issue.

bhilbert4, 2022-03-25:
Hopefully not. The web app doesn't need to fool with fits files when rendering pages. But I guess if someone tries to open the web app while e.g. the dark monitor is running....who knows? 

#867 will hopefully help some with app slowness.

bhilbert4, 2022-03-29:
I agree that something like this would be helpful. See #907 (although there isn't much there at the moment besides "we need something different").

PR #908 (2022-06-10): Reduce memory usage by the dark monitor
This PR makes adjustments to the dark current monitor in order to try and reduce the overall memory usage. I'm afraid there isn't much we can do here. The highest memory usage comes when running the cal pipeline on the input files. 

One option would be to skip the cal pipeline and fall back to using (last group - first group) as an approximation of the slope image. 

- [ ] ~~Look into what effects using the difference image would make.~~ 
- [ ] ~~Would CRs throw off the statistics since we'd be skipping the jump step~~
- [ ] ~~Would the dark rate histog [... cut]
PR #921 (2022-10-21): Celery monitors
This will alter the existing monitors (as little as possible) by replacing the calls to the pipeline_tools calibration function with a blocking call to the celery calibration task. As such, code flow will work exactly as it would before, except that (of course) instead of waiting for the pipeline to run, the code will instead wait for the pipeline task to move up the queue, and then run.

Closes #905 
Closes #907