spacetelescope/jwql #907: Shared pipeline outputs for instrument monitors

bhilbert4, 2022-03-24:
Currently each instrument monitor works in its own silo. Data are retreived from MAST, steps of the pipeline may be run, and analyses are done.

It could be helpful to have a given monitor check the output products from other monitors to see if a file it needs already exists in the JWQL products area. In that case, we could save having to run pipeline steps, which would save time as well as memory. With more and more monitors being added, there will be times when more than one monitor is running at a time on the server. We've seen that with large input files (and most likely multiple monitors running) we can use up all available memory. (#905)

Perhaps a separate module that can search all monitor output directories for the version of the file needed, and then creates a symbolic link to it if it exists?

bhilbert4, 2022-03-29:
Perhaps a centralized module that knows for data with a particular instrument/aperture/exp_type/etc, the dark monitor needs steps x, y, z of the pipeline run, and outputs from steps x and z, while the bad pixel monitor needs steps w, x, y, z run, and outputs from w and z. And then it will run those steps and save the outputs at the needed steps, and perhaps create symbolic links for the appropriate files to each monitor's output directory.

bhilbert4, 2022-03-31:
I think it would be interesting to see. How would the reduction server know which pipeline steps to create outputs for? For example, if both the dark monitor and the readnoise monitor will use the same file, but the dark monitor needs the output from the jump step and the ramp-fitting step, while the readnoise monitor needs the output from the refpix subtraction step. It seems like we would need a mapping somewhere to describe this, since the dark monitor itself, when asking for a calibrated file, doesn't know anything about the calibrations needed by other monitors.

york-stsci, 2022-03-31:
I guess one thing that I hadn't thought of until now is what happens if monitor A needs file X, and monitor B also needs file X, and B submits the request while the request from A is running, but before it finishes. If we only have the resources to run a single worker thread, that's not a problem, because by the time monitor B's task runs the file will exist, but if we're running in parallel we have to have some way to check for whether a particular calibration is already ongoing. That said, I suspect that this is *also* a solved problem somewhere, and we can borrow or create our own solution if we're ever in the fortunate place of being able to run multiple pipeline threads simultaneously.

bhilbert4, 2022-03-31:
I think that does make sense, and could be really useful. Handling the pipeline outputs might be a little tricky. By default, the pipeline saves the output only from the ramp-fitting step, which is the final step in stage 1 of the pipeline. (I think we can ignore stages 2 and 3 of the pipeline for now). In order to save the results from any other step, the user has to specify that. 

With my NIRCam hat on, the steps of the stage 1 pipeline are:
1. Attach bad pixel mask to the data
2. Flag saturated pixels
3. Subtract superbias
4. Subtract reference pixels
5. Linearize the ramp
6. Dark subtraction
7. Flag CR hits
8. Ramp-fitting

So if the dark monitor needs the output from step 7, while the readnoise monitor needs the output from step 4, in order to get both of those outputs from a single run of the pipeline, you need to specify those outputs in the pipeline call. If you were to ask only for the output of step 7, there won't be an output file saved from e.g. step 3 that can be picked up and have step 4 run on it later. You would have to start over from the beginning.

bhilbert4, 2022-03-31:
I did create some tools that sound similar, in https://github.com/spacetelescope/jwql/blob/develop/jwql/instrument_monitors/pipeline_tools.py

In that case the code checks what pipeline steps have been run on a given file and determines what steps need to be run in order to get the desired output. It uses boolean tables like you mentioned. This was meant to save time in terms of not running and re-running the same initial pipeline steps. In practice I'm not sure how useful it actually is, since the starting files from MAST will usually be uncal (i.e. raw). For dark current files, there is also a *_dark.fits file in MAST, which has been run through the first 5 steps above. In that case I think the tool helps for cases where we also want the output from steps 7 and 8.

bhilbert4, 2022-04-01:
I'm betting that in almost all cases we will never want the output from all the pipeline stages. But we may be able to narrow things down to the outputs from just a few steps.

york-stsci, 2022-04-01:
Okay.

Based on my testing so far, it seems like the lowest-disruption way of doing this is to create a celery pipeline task (probably using the redis backend, just because that's what I'm familiar with), and including celery-singleton (which basically will look for identical requests to calibrate the same file and, instead of starting another calibration task, return the AsyncResult that corresponds to the existing task.

At that point, the lowest-disruption way of working on the monitors is to take the part of the monitor that calibrates data, and replace it with

`result = calibration_task.delay(file_to_calibrate)`
`path_to_calibrated_data = result.get()`

which will act as a blocking call that will just delay until the calibrated data is available. There are also ways to not block on waiting for a result, and instead include a callback (or equivalent), to let the monitor queue up a bunch of reductions and then idle while waiting for them to finish, but that would require more changes to the monitors, and the two methods can co-exist quite easily, so we can make the changes when and as needed.

PR #921 (2022-10-21): Celery monitors
This will alter the existing monitors (as little as possible) by replacing the calls to the pipeline_tools calibration function with a blocking call to the celery calibration task. As such, code flow will work exactly as it would before, except that (of course) instead of waiting for the pipeline to run, the code will instead wait for the pipeline task to move up the queue, and then run.

Closes #905 
Closes #907
closed 2022-10-21 by york-stsci