spacetelescope/jwql #914: How should we treate guide star observations?

bhilbert4, 2022-03-31:
Currently, generate_preview_image.py creates preview images for guide star observations (e.g. jw01602002001_gs-acq1_xxxx.fits).  This means that these files are in the JWQL filesystem. However, these files don't show up on the page for a given program, so these preview images and associated thumbnails are not actually used at all.

So the question is, should we skip over making preview images for guide star products? Or should we add guide star products to the web app (i.e. make them visible/inspectable)? There are a ton of files. I think if we were to add them, we'd want to find a way to keep them separate from the science data, just to avoid confusion and make things easier to read/scroll.

Here are the file types we are currently making preview images for:
gs-acq1
gs-acq2
gs-track
gs-fg

@shanosborne any thoughts? Would the FGS team use something like this? Do you think JWST users would want the ability to look at the guide star data for their observations?

shanosborne, 2022-04-01:
Great question, I will raise it to the FGS team and see what they think.

shanosborne, 2022-04-04:
@bhilbert4 The FGS team thinks they would definitely use something like this and also that JWST users could want to use it too for cases where the science data looks off (e.g. drifting). So they would like this functionality.

However, they would like to also have access to the gs-id data along with the acq, trk, and fg. Do you know why the ID data isn't already available?

bhilbert4, 2024-05-13:
For the moment, JWQL ignores guide star data. If we decide to include it in the future, the [spelunker](https://github.com/GalagaBits/JWST-FGS-Spelunker) package would be a good place to start.

closed 2024-05-13 by bhilbert4