spacetelescope/jwql #926: Clean up preview images and thumbnails in test data set

bhilbert4, 2022-04-14:
It looks like we're missing some thumbnail images, and have others that are outdated (e.g. "mosaic" thumbnails) in the test data location on the central store. It would be worth removing the outdated images, and re-running generate_preview_images.py there, in order to be sure we have a complete dataset that can be used for code development.
This is in the ins/jwql/thumbnails subdirectory on the central store. See a local config.json file for the full path.

cracraft, 2022-04-14:
Does this include the thumbnails that look like axes of a plot?

bhilbert4, 2022-04-14:
No, this is for our test data, rather than the flight data. I'll make sure we have a separate issue for the issue with the weird plot thumbnails.

bhilbert4, 2024-05-13:
This work is complete. Test datasets have been moved onto the central store.

closed 2024-05-13 by bhilbert4