spacetelescope/jwql #928: Some preview images contain plot axes

bhilbert4, 2022-04-14:
In some cases, preview images/thumbnail images appear to show plot axes, rather than a simple image of the data. @cracraft do you have any specific examples?

bhilbert4, 2022-04-14:
Looks like NIRCam 1062, 1073, 1162 are examples. I also see mosaicked images being used as thumbnails in some other cases. We shouldn't be making mosaic preview images/thumbnails anymore, so I'm not sure how that can be.

cracraft, 2022-04-14:
MIRI proposals 01546, 01030, and 01012 (has what looks like data in the plots) all have examples of this.

BradleySappington, 2022-04-25:
In order to re-create a thumbnail, BOTH the preview_image AND the thumbnail cannot exist.
Since all thumbnails/preview images follow a specific naming convention, we can create a file to house image names that need to be re-processed.

Then when we run the command (which I think happens daily), It can delete the thumbnail/preview_image files and re-generate them.  Once used, the file will be emptied.

BradleySappington, 2022-04-26:
After Discussion with @bhilbert4 and @mfixstsci, we decided to manually delete the problematic files before updating the code base.  When our program re-runs, it will shed light on how the image corruptions are occurring.

BradleySappington, 2022-05-02:
Currently previews and thumbnails are being created from sources that should be excluded.  Pausing this effort until the proper image sources are used.

bhilbert4, 2022-05-03:
I spent some time looking at the calls to generate_preivew_images, and I see that we're using multiprocessing there. I wonder if that's our problem. Perhaps with multiple figures open at the same time, information is moving across instances somehow, and contaminating some plots? With Mirage, I've seen situations (especially starting in python 3.8), where multiprocessing threads seem to become nested or cross-contaminated. It might be worth making a test run of generate preview images without multiprocessing and see if the problem goes away. A while back I read about changes to the way multiprocessing works starting in python 3.8. Spawning vs. forking was part of it. https://britishgeologicalsurvey.github.io/science/python-forking-vs-spawn/

closed 2022-05-16 by mfixstsci