spacetelescope/jwql #956: More incorrect file counts on Archive pages

bhilbert4, 2022-05-20:
NIRCam program 1410 is listed as having 1295 files. But in displaying the program files one observation at a time, I'm counting a total of only ~85 files or so. Either not all of the files are being displayed on the program pages, or the total file count is incorrect.

bhilbert4, 2022-05-20:
It looks like perhaps #948 is not finding the complete list of obsnums for some programs. I just checked 1166, which is listed as having 445 files. The drop down menu wth the list of obsnums is missing quite a few entries when I compare it with the APT file.

stscijgbot-jwql, 2024-05-13:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-228?focusedCommentId=698790#comment-698790):_

It looks like perhaps #&#x2060;948 is not finding the complete list of obsnums for some programs. I just checked 1166, which is listed as having 445 files. The drop down menu wth the list of obsnums is missing quite a few entries when I compare it with the APT file.