spacetelescope/jwql #989: Show plots from x1d files

bhilbert4, 2022-07-18:
Would it be worth adding code to be able to read and plot the data contained in x1d files?

mengesser, 2022-07-26:
I think the MIRI team expressed that this is desirable. There are some effects that are only visible when you look at the actual spectrum, and being able to review them would be great.

stscijgbot-jwql, 2024-08-22:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-225?focusedCommentId=714781#comment-714781):_

Another vote for this today at the JWQL presentation to JWST Project Scientists.
It seems like this could be relatively straight forward for MIRI/NIRSpec. For the NIRCam and NIRISS WFSS modes, dealing with the many-to-one relation between level 2 image and level 3 spectra will have to be considered.