spacetelescope/jwql #1014: Display level 3 images

bhilbert4, 2022-07-28:
It would be nice to include level 3 imaging products in JWQL (creating/showing preview images). Currently this is not done. Level 3 files are present in the 'L3' subdirectory for each program in the filesystem. However, level 3 files are not detector-specific (and perhaps not even observation specific?), so they'll have to be treated differently than the level 1 and 2 files we are currently displaying)

bhilbert4, 2022-12-06:
Another vote for this today in the NIRCam meeting.

This will take some thought, because there is a large diversity of level 3 products. 
Maybe skip the non-image products (e.g. 1d spectra) for the time being? 

With imaging, we'd have to think about the metadata associated with the products. For example, when mosaics are created from multiple detectors, we'll need to do something so that the information shown in the hover over is useful/not an error.

stscijgbot-jwql, 2024-05-13:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-223?focusedCommentId=698782#comment-698782):_

Another vote for this today in the NIRCam meeting.

This will take some thought, because there is a large diversity of level 3 products.
Maybe skip the non-image products (e.g. 1d spectra) for the time being?

With imaging, we'd have to think about the metadata associated with the products. For example, when mosaics are created from multiple detectors, we'll need to do something so that the information shown in the hover over is useful/not an error.

stscijgbot-jwql, 2024-08-22:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-223?focusedCommentId=714782#comment-714782):_

There was some more discussion of this today at the JWQL presentation to JWST Project Scientists, in concert with [JWQL-225](https://jira.stsci.edu/browse/JWQL-225). Being able to see Level 3 products can help to diagnose calibration issues that are not apparent in the level 1 and 2 images. e.g. WCS issues, odd-looking spectral features, etc.