spacetelescope/jwql #1017: When vao.stsci.edu (or other) service is unavailable, alert user instead of hanging forever

BradleySappington, 2022-08-02:
**Describe the bug**
When the site was down, the magnifying glass on the archive page went on forever.  I had to run locally to discover the call stack trace issue.

```
Traceback (most recent call last):
  File "/Users/bsappington/miniconda3/envs/jwql-3.8/lib/python3.8/site-packages/django/core/handlers/exception.py", line 47, in inner
    response = get_response(request)
  File "/Users/bsappington/miniconda3/envs/jwql-3.8/lib/python3.8/site-packages/django/core/handlers/base.py", line 181, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
  File "/Users/bsappington/repositories/jwql/jwql/website/apps/jwql/views.py", line 276, in archived_proposals_ajax
    all_proposals = get_instrument_proposals(inst)
  File "/Users/bsappington/repositories/jwql/jwql/website/apps/jwql/data_containers.py", line 864, in get_instrument_proposals
    tap_results = tap_service.search(f"select distinct proposal_id from dbo.ObsPointing where obs_collection='JWST' and calib_level>0 and instrument_name like '{instrument.lower()}'")
  File "/Users/bsappington/miniconda3/envs/jwql-3.8/lib/python3.8/site-packages/pyvo/dal/tap.py", line 246, in run_sync
    return self.create_query(
  File "/Users/bsappington/miniconda3/envs/jwql-3.8/lib/python3.8/site-packages/pyvo/dal/tap.py", line 942, in execute
    return TAPResults(self.execute_votable(), url=self.queryurl, session=self._session)
  File "/Users/bsappington/miniconda3/envs/jwql-3.8/lib/python3.8/site-packages/pyvo/dal/query.py", line 244, in execute_votable
    self.raise_if_error()
  File "/Users/bsappington/miniconda3/envs/jwql-3.8/lib/python3.8/site-packages/pyvo/dal/query.py", line 253, in raise_if_error
    raise DALServiceError.from_except(e, self.queryurl)
pyvo.dal.exceptions.DALServiceError: 503 Server Error: Service Unavailable for url: http://vao.stsci.edu/caomtap/tapservice.aspx/sync
[02/Aug/2022 09:58:20] "GET /ajax/FGS/archive/ HTTP/1.1" 500 105432
```

**To Reproduce**
occurred when vao.stsci.edu/ was down for maintenance

**Expected behavior**
If a service is unavailable in vao.stsci.edu/, we should alert the user that a specific site is down instead of just hanging.
This will allow them to raise a ticket (or ask us to do it).

**Extra Information**
This particular issue occured
File "/Users/bsappington/repositories/jwql/jwql/website/apps/jwql/data_containers.py", line 864, in get_instrument_proposals
    tap_results = tap_service.search(f"select distinct proposal_id from dbo.ObsPointing where obs_collection='JWST' and calib_level>0 and instrument_name like '{instrument.lower()}'")

bhilbert4, 2024-05-13:
@BradleySappington I'm assuming this is no longer an issue since we are now running with debug=False?
Is there any work that needs to happen here, or should we close this?

BradleySappington, 2024-05-13:
I don't think debug mode would affect this at all.  This looks like an ajax call didn't have a raised exception handled properly.  This should definitely at least be investigated again.

stscijgbot-jwql, 2024-05-13:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-229?focusedCommentId=698791#comment-698791):_

This needs to be investigated further.