spacetelescope/jwql #1072: header keywords to appear when hovering over thumbnails

penaguerrero, 2022-10-31:
the NIRSpec team would like these header keywords to show when hovering over the thumbnails instead of the current info:
- EXP_TYPE
- DETECTOR
- FILTER
- GRATTING
- PATT_NUM
- EXP_START
- LAMP
- OPMODE

bhilbert4, 2022-10-31:
I agree that Proposal and Observation could be discarded, since all thumbnails on a given page will have the same values for those. Keeping visit could be helpful in terms of easily placing a given file within an APT observation. (Or am I the only one who uses visit numbers? It could very well be the case.) For NIRCam at least, some note on detector would also be helpful to keep. Do you think people would be ok with removing the entry for the observation date?

Also, @rcooper295 is working on updates to the hover-over info. See #1019 We should probably incorporate any updates from this issue into that as well.

cracraft, 2022-11-01:
Can we have the information in the hover over be different for each instrument? This list isn't necessarily relevant to the MIRI team, but I can ask what they'd like to see. I can put that information in the issue #1019 as listed above.