spacetelescope/jwql #1103: CR monitor improvements

bhilbert4, 2022-11-15:
At the MIRI team meeting today, the general consensus was that the CR monitor is not currently doing what the team wants. 

The raised value of the threshold value in the call to the jump step is not wanted.
There was debate about whether it's useful to capture the jump magnitudes at all.
There are plans to ask DMS to save the jump step output as a standard product, at which point the monitor won't need to call the pipeline. 
There are plans to add the number of snowballs/CR showers to the header of each file.

Given this, it seems like we may want to stop the current run of the CR monitor and not run it until the team(s) decide what information they want the monitor to produce and save.

bhilbert4, 2022-11-15:
Waiting on a response from Karl Gordon and Mike Regan on whether the current CR monitor outputs are at all useful.

bhilbert4, 2022-11-28:
Karl and Mike are going to make some thorough revisions to the CR monitor. The monitor has been shut down until the updated code is in place.

bhilbert4, 2023-07-05:
New header keywords are now in science data, which specify the number of CRs found in each file. See [JP-3096](https://jira.stsci.edu/projects/JP/issues/JP-3096) These were added in 9.3RC2, which is jwst version 1.11.1. 

This means that new data starting from ~July 1 will have these keywords in the headers. We can now begin work to edit the CR monitor to read in these keyword values and produce plots.

stscijgbot-jwql, 2024-05-13:
_Comment by [Bryan Hilbert](https://jira.stsci.edu/secure/ViewProfile.jspa?name=hilbert) on [JIRA](https://jira.stsci.edu/browse/JWQL-39?focusedCommentId=698745#comment-698745):_

CR information is now stored in file headers. That would make this monitor relatively simple, assuming all we would want to do is plot the time evolution of CR rates.