spacetelescope/jwql #1106: Optionally show multiple detectors on an exposure page

bhilbert4, 2022-11-15:
The NIRSpec team requested the ability to view the two detectors side by side for an exposure. This will help when examining traces that span multiple detectors.

Similarly for NIRCam, this could be helpful for viewing extended sources that span multiple detectors.

There are two options for this that come to mind at the moment:
1. Implement a button that the user can toggle. When pressed, the preview images for neighboring detectors are shown next to the preview image for the current detector.
2. We re-implement support for level 3 pipeline outputs (at least for imaging mode). In this case, we make preview images for i2d files, and add them to the list of suffixes. This method has the advantage that the JWQL team doesn't need to think about relative detector locations, and doesn't require significant changes to the exposure-level webpages.

For modes other than imaging, we may not want to support i2d files yet, since in many cases those files contain (possibly lots of) 1-d tables of spectra.

melanieclarke, 2023-02-23:
@bhilbert4 - I think I mostly understand what's needed here and have an idea on how to implement it.  Can you please check my understanding and let me know if this is what you had in mind?

Here's the user story:

Exposure groups are files that share proposal ID, observation number, visit number, activity number, exposure number, and segment ID (if applicable).  In effect, the root name up to but not including the detector must match.

On the observation archive view ({instrument}/archive/{pid}/obs{obsnum}), we add a checkbox to display thumbnails by exposure groups instead of by root file.  This box is checked by default.

When the group button is unchecked, the observation view is the same as it is currently: individual root files have their own thumbnails and clicking on them takes you to that file's image page ({instrument}/{rootfile}).  From the image page, you can select new/viewed status, change file type, view header, explore image, or mark an anomaly.  Previous/next buttons take you to the previous/next image, following previously set filter and sort rules.

When the group button is checked, the observation view shows thumbnails by exposure group.  Clicking on one takes you to an exposure group page ({instrument}/{exposure_group}).  In this page, images from all available detectors are displayed, with relative placement approximating on-sky locations if possible.  Clicking on an image takes you to that root file's image page, as above.  From the group page, you can:
•	Select new/viewed for the exposure group.  The status is applied to all files in the group.
•	Choose a new file type to display. The selection is applied to all displayed images.
•	Mark anomalies to apply to the whole group.  To mark an anomaly for a specific image in the set, click through to its image page and select it there instead.
•	View header or explore image for one of the displayed files, selected by a dropdown box.
Previous/next buttons on the group page take you to the previous/next exposure group, following previously set filter and sort rules.

For instruments or observations that do not have exposure groups containing multiple detectors, the observation archive view behaves the same as it does currently.  In this case, the group checkbox has no effect.

bhilbert4, 2023-02-23:
That sounds amazing! Reviewing by exposure group rather than by rootname sounds like exactly what users would want and should save a ton of time. 

The complicated case here is NIRCam. 10 detectors, but all 10 are not always read out. There are cases where only 5 are read out (NRCA* or NRCB*), cases where only 3 are read out (A1, A3, A5 or B2, B4, B5), and cases (subarrays) where 2 are read out (B1, B5). There may be a case where only 1 is read out. I forget if we get data from more than 1 detector for target acquisition images.

NIRSpec and FGS should be a lot easier, with only 2 detectors each, where you can get data from both, or just one.

melanieclarke, 2023-02-23:
Yes, the display rules for NIRCam will probably take some iteration.  Do you have a source you can point me to for the possible configurations?

For FGS, I noticed that there are guider1 and guider2 images, but I haven't found a case where there are both in the same observation.  Do you know of an example where both are taken at the same time?

bhilbert4, 2023-02-23:
I'll look around. For NIRCam, I'm not sure there's one place that all the configurations are spelled out. In the meantime, Figure 1 [here](https://jwst-docs.stsci.edu/jwst-near-infrared-camera/nircam-instrumentation/nircam-detector-overview) shows the layout of the instrument. 

For FGS there is an option to guide with one detector while obtaining science data with the other. I think it's rarely used, but I did hear about a program in commissioning that tried it. I'll dig around to see if I can find it. It may be in that case that the guiding data are not put in the same exposure group as the science data.

melanieclarke, 2023-02-23:
Okay, thank you!  It sounds like I've got the principles right anyway, so I'll get started on the structure and sort out the instrument specifics later.

closed 2023-03-20 by mfixstsci