spacetelescope/jwql #1165: Web app stability vs quick access to new features

bhilbert4, 2023-02-02:
The production server is still currently pointed at the ```develop``` branch. This is dangerous in cases where updates merged into ```develop``` break things. But at the same time, teams want to be able to use the latest features as soon as they are released. How can we reconcile this?

Here are two options:

1) Use one of our servers as a new "QA" server. Have this server host the web app (e.g. jwql-qa.stsci.edu), pointing to the ```develop``` branch (exactly what the production server is doing now). Change the production server so that it is pointing to the ```main``` branch. Then teams who want the latest features can use the QA server to test things out, with the understanding that sometimes the new features may break things. But the production server will always be available, even if it doesn't support the latest features. Once features are well tested, they are merged from the ```develop``` branch into the ```main``` branch, at which point they will be available on the production server.

2) Point the production server to the ```main``` branch. New feature testing will still be done (by the JWQL team only since no one else has access) on the test server. Increase the cadence of minor releases such that the latest features are quickly brought over to the production server once they are deemed correct.

bhilbert4, 2023-02-02:
For option 1, how would we deal with databases? We wouldn't want new features on the QA server to change the production database, in case there is some problem with a new feature. So the QA server would need to point to a new copy of the database? the test database? Once the new features are merged from ```develop``` into ```main```, we'll also potentially need to migrate any changes from the QA database into the production database.

This option will require ITSD to modify one of our servers such that it can be reached via a new jwql-xxxx.stsci.edu URL, and use the same permission settings that the production server uses, so that instrument team members can access the site.

bhilbert4, 2023-02-02:
For option 2, no new technical changes are needed. Instead we would need to adjust the team policy on testing new features on the test server to be done in a more rigorous, well-defined way, in order to catch potential side effects/problems. (e.g. regression tests for webpages?) We would also need to make releases more frequently, in order to get new features out to the teams as quickly as practical.

bhilbert4, 2023-02-02:
@BradleySappington @york-stsci @mfixstsci I've put down some initial notes, including an alternative situation to what was discussed in the meeting. Feel free to add thoughts and ideas.

BradleySappington, 2023-02-02:
2 different databases can be challenging.  Because if folks updated main, it wont be updated in QA.  So if folks are using QA as their primary access to specific jwql data, it will not be complete.  For example if someone filtered anomalies on QA and DEV they'd get different results.

This makes me think that if we move forward with QA, we'd probably have to have a little note there saying all DB updates should be made on PROD. or something along those lines.

york-stsci, 2023-02-02:
The problem is if we run into something like adding a new anomaly type, and so the only way of adding it is to QA. I'm not sure what the best way of dealing with that is, but it's possible we might have to do something like a shared database where you can add new columns (and reference them on QA), but you can't remove columns, or change their formats, except as part of a general update. Which doesn't sound ideal, but which may be what we need to do.

My current thought is that probably we only want to do this for user-populated tables (and have completely separate monitor-populated tables), because that way we can have separate monitor scripts (which is going to probably lead to some kind of annoying issues, but I don't know offhand what those will be).

bhilbert4, 2024-05-10:
This issue has been resolved with our new release cadence combined with following best practices with our server use.

closed 2024-05-10 by bhilbert4