spacetelescope/jwql #1178: Make the query page fully functional

bhilbert4, 2023-02-13:
Update the query page such that users can filter and search for data using various observation/exposure properties. 

As part of this, move more information about exposures into the Django models, in order to make filtering easier.

BradleySappington, 2023-02-13:
- [ ] Have Info in Thumbnails
- [ ] create datamodel entries to hold all jwql_query fields 
- [ ] update datamodels with query information from fits header
- [ ] update datamodels with query anomaly information from postgres DB
- [ ] verify all thumbnails are returned on query

BradleySappington, 2023-02-13:
Here are the options for the filter keyword in the fits files. This list is a mix of the filters across all the instruments: https://github.com/spacetelescope/stdatamodels/blob/master/src/stdatamodels/jwst/datamodels/schemas/keyword_filter.schema.yaml

Here are the pupil values, which apply to NIRCam and NIRISS: https://github.com/spacetelescope/stdatamodels/blob/master/src/stdatamodels/jwst/datamodels/schemas/keyword_pupil.schema.yaml

Gratings, for NIRSpec: https://github.com/spacetelescope/stdatamodels/blob/master/src/stdatamodels/jwst/datamodels/schemas/keyword_grating.schema.yaml

BradleySappington, 2023-03-14:
TODO list when merging all to main:

when merged into development and production the following will need to occur on the dev server, and the migrations should be merged to the production server:

$ python manage.py makemigrations
$ python manage.py migrate

We will need to temporarily disable the cronjobs on production for archive_database_update.py
Once halted we will need to run manually with the new update flag.

$ python archive_database_update.py update
this will take a few hours.
when complete enable cronjobs again

populate the anomalies in the models:
$python manage.py makemigrations
$python manage.py migrate
$python anomaly_db_transfer.py

BradleySappington, 2023-03-29:
#1229 was needed to be in place before this work could continue

closed 2023-04-25 by mfixstsci