spacetelescope/jwql #1185: create datamodel entries to hold all jwql_query fields

bhilbert4, 2023-02-14:


melanieclarke, 2023-02-22:
@BradleySappington - can we please add exp_type to the RootFileInfo model as well?  There is a list of exptypes in Observation already, but I'm realizing that "observation" is a broader concept than I originally thought and it would be useful to have the specific exp_type for each root file as well.

BradleySappington, 2023-02-22:
@melanieclarke, yeah good catch, its a list in the observation so wont do much good unless its held at the base level.

BradleySappington, 2023-03-03:
when merged into development and production the following will need to occur on the dev server, and the migrations should be merged to the production server:

$ python manage.py makemigrations
$ python manage.py migrate

PR #1194 (2023-03-13): Update Models for Filter Fields
closes #1185