spacetelescope/jwql #1207: Mostly white pixels on preview image and thumbnail

hover2pi, 2023-03-02:
**Describe the bug**
For some NIRISS '_rate.fits' images with high background flux (e.g. https://jwql.stsci.edu/NIRISS/jw01208034001_02101_00001_nis/), the preview image and the thumbnail shows mostly white pixels that make the image look like a failed observation. However, inspecting the image shows the white pixels are not present. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://jwql.stsci.edu/NIRISS/jw01208034001_02101_00001_nis/
2. Click on the radio button for 'rate' file type to see the white pixels.
3. Click on the 'Explore Image' button to see that the pixels aren't white after all but the background is high.

**Expected behavior**
The preview image should look the same as the full image viewed after clicking the 'Explore Image' button

**Additional context**
The background in the affected images are 10 - 25% of the signal flux, pretty high but not unusable.

bhilbert4, 2023-03-07:
I'm seeing this in a few other cases, all with data taken very recently: 
https://jwql.stsci.edu/NIRCam/archive/2514/obs011/
https://jwql.stsci.edu/NIRCam/archive/1783/obs001/
https://jwql.stsci.edu/MIRI/archive/1783/obs002/

bhilbert4, 2023-03-07:
And more:
https://jwql.stsci.edu/NIRSpec/jw01491027001_01201_00001_nrs1/

I wonder if this is related to the dependency updates that were made recently. Maybe a color palette has changed, or something like that?

bhilbert4, 2023-03-07:
Looking more closely at the data in ds9, I'm thinking this is due to the recent change in the pipeline where bad pixels are given a value of NaN in the rate image. Prior to this change, these pixels were given a value of zero. I think that when the preview images are being created, these NaN pixels, which are white in the current color palette, are not disappearing as well as they did previously when matplotlib bins/averages/downsamples the pixels during the creation of the jpg. 

So maybe we need a new step in the creation of the preview images that replaces NaN values with local averages before saving the jpgs? Or return to the previous situation and set them equal to zero?

PR #1212 (2023-03-13): Update preview image generator to handle new NaN pixels
Resolves #1207 

A recent change in the cal pipeline now has pixels for which no slope is computed being set to NaN in the slope image. Previously these pixels had been set to zero. Setting these pixels to NaN means matplotlib now shows them as white in the preview images rather than black. This, combined with the way that matplotlib averages pixels together when producing the preview image/thumbnail jpgs was resulting in preview images that were full of white splotches. This PR updates the preview image generator such that NaN pixels are [... cut]
closed 2023-03-13 by mfixstsci